jkjung-avt/tensorrt_demos #173: TensorRT YOLOv3 For Custom Trained Models，tensor RT infers that there is no rectangular box, conf_ Th = 0.001 ~ 0.3 without frame?

honghande, 2020-08-01:
The height and width of the trained yolov3.cfg are both 416. Should the inference input of the trained model be set to 416? If this is the case, there are only four categories in my category, and the input h and W are set to 416, but the inference result is that there is no rectangular box and no category result. The suffix generated by the model I trained is in PT format, and the result can be detected in the same image. What's the matter?

jkjung-avt, 2020-08-02:
Please verify your model with darknet first.

honghande, 2020-08-02:
@jkjung-avt Has verified the results。Thank you！
I am using the pytorch version of yolov3. I used the model trained with my own data to test the same image and there are results. Now I use this demo to convert to onnx successfully and no error is reported. Then I transfer to tensorrt successfully and no error is reported. Use the converted trt file Inferred, the detection output is [],[],[]

jkjung-avt, 2020-08-02:
The code in this repository only converts darknet cfg/weights to onnx.  I myself have not tested pytorch implementation of yolov3.  I cannot guarantee the code would work OK for pytorch yolov3 models.

honghande, 2020-08-02:
@jkjung-avt This code is https://github.com/ultralytics/yolov3, I did not fail to convert onnx, nor did I fail to convert tensorrt, you can try the conversion if you have time, thank you!
I have another question to ask you. The height and width in yolov3.cfg are the size of the image we need to resize, right?

jkjung-avt, 2020-08-03:
> I have another question to ask you. The height and width in yolov3.cfg are the size of the image we need to resize, right?

Yes.

Just for reference.  The resizing is done in my code here: https://github.com/jkjung-avt/tensorrt_demos/blob/a0cd57c62daf6fe1b08f120dc2bffe29f660d62d/utils/yolo.py#L60

closed 2020-08-07 by jkjung-avt