jkjung-avt/tensorrt_demos #176: python3: nmsPlugin.cpp:54: virtual nvinfer1::Dims nvinfer1::plugin::DetectionOutput::getOutputDimensions(int, const nvinfer1::Dims*, int): Assertion `nbInputDims == 3' failed.

MarsJacobs, 2020-08-04:
Hi! Thanks for your help.

we tried to make  .UFF file using build_engine.py with other SSD model **ssdlite_coco** (from Model Zoo).

following is our environment:

Jetson Nano
Jetpack : 4.3
TensorFlow : 1.15.2

And this is the ERROR message

python3 build_engine.py ssdlite_coco
2020-08-04 14:50:52.428760: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library libcudart.so.10.0
[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace: 
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/graphsurgeon/StaticGraph.py:125: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING: To create TensorRT plugin nodes, please use the `create_plugin_node` function instead.
NOTE: UFF has been tested with TensorFlow 1.14.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
UFF Version 0.6.5
=== Automatically deduced input nodes ===
[name: "Input"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_FLOAT
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: 1
      }
      dim {
        size: 3
      }
      dim {
        size: 300
      }
      dim {
        size: 300
      }
    }
  }
}
]
=========================================

Using output node NMS
Converting to UFF graph
Warning: No conversion function registered for layer: NMS_TRT yet.
Converting NMS as custom op: NMS_TRT
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:179: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.

Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_conf as custom op: FlattenConcat_TRT
Warning: No conversion function registered for layer: FlattenConcat_TRT yet.
Converting concat_box_loc as custom op: FlattenConcat_TRT
DEBUG [/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/converter.py:96] Marking ['NMS'] as outputs
No. nodes: 589
UFF Output written to /home/imzab/project/tensorrt_demos/ssd/tmp_v2_coco.uff
UFF Text Output written to /home/imzab/project/tensorrt_demos/ssd/tmp_v2_coco.pbtxt
python3: nmsPlugin.cpp:54: virtual nvinfer1::Dims nvinfer1::plugin::DetectionOutput::getOutputDimensions(int, const nvinfer1::Dims*, int): Assertion `nbInputDims == 3' failed.
Aborted (core dumped)

And this is the part of the build_engine.py we added

 'ssdlite_coco' : {
        'input_pb':   os.path.abspath(os.path.join(
                          DIR_NAME, 'ssdlite_coco.pb')),
        'tmp_uff':    os.path.abspath(os.path.join(
                          DIR_NAME, 'tmp_ssdlite_coco.uff')),
        'output_bin': os.path.abspath(os.path.join(
                          DIR_NAME, 'TRT_ssdlite_coco.bin')),
        'num_classes': 91,
        'min_size': 0.2,
        'max_size': 0.95,
        'input_order': [0,2,1],  # order of loc_data, conf_data, priorbox_data
    },

How can we fix this Error? Thank you

MarsJacobs, 2020-08-04:
> Hi! Thanks for your help.
>
> we tried to make .UFF file using build_engine.py with other SSD model **ssdlite_coco** (from Model Zoo).
>
> following is our environment:
>
> Jetson Nano
> Jetpack : 4.3 [...]

joelosw, 2020-08-05:
Without investigating this thoroughly I would guess that you input orders (to the NMS plugin) are set wrong.

MarsJacobs, 2020-08-06:
@joelosw Thanks for your answer! How can I change the input order to the NMS plugin?

joelosw, 2020-08-06:
1. Have a look at the .pbtxt file that was created. 
In the beginning you should find a node "NMS" with three inputs. 
2. Write down/remember the input order
3. In the build_engine.py there is a section MODEL_SPECS. I suppose you added your own section for your model. One of the specs is the input order. Insert the previously looked up position of Squeeze/loc_data, conf_data,priorbox_data

jkjung-avt, 2020-08-07:
This appears to be the same issue as: https://github.com/jkjung-avt/tensorrt_demos/issues/32

The assertion failure seems to happen on this line of code (although the line number does not match): https://github.com/NVIDIA/TensorRT/blob/release/6.0/plugin/nmsPlugin/nmsPlugin.cpp#L82

I think it means the UFF graph is incorrect.  The NMS_TRT plugin expects exactly 3 inputs, but in the UFF graph there are either too few or too many input nodes (most likely too many) to the NMS node.  So you need to debug the UFF graph, find the redundant input(s) to NMS, and probably use graphsurgeon to remove it.

This discussion on NVIDIA Developer Forum might be helpful: https://forums.developer.nvidia.com/t/sampleuffssd-with-custom-ssd-mobilenet-v1-model/70508

closed 2020-08-18 by jkjung-avt