jkjung-avt/tensorrt_demos #179: Problems with mtcnn？

jiagh2010, 2020-08-04:
Hi,
Run successfully at the terminal‘ trt_mtcnn.py ’.
But I can't use mtcnn. The procedures and problems are as follows：
codes：

    mtcnn = TrtMtcnn()
    '''
    path1="./data/"
    i=0
    for filename in os.listdir(path1):
        path=path1+filename
        print(i,path)

        img_cv=cv2.imread(path)
        img_cv=cv2.resize(img_cv,(640,480)) 
        dets, landmarks = mtcnn.detect(img_cv, minsize=40)
        print('{} face(s) found'.format(len(dets)))
        img = show_faces(img_cv, dets, landmarks)
        cv2.imwrite("./result/"+str(i)+".jpg",img) 
        i=i+1
    '''
    #del(mtcnn)

problems：

Deserialize required 1226718 microseconds.
Current optimization profile is: 0. Please ensure there are no enqueued operations pending in this context prior to switching profiles
Assertion fail in file 'trtNet.cpp' line 186: bad dims for 'data'
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted (core dumped)

jkjung-avt, 2020-08-07:
Actually, you don't need to resize the image before calling mtcnn.detect().  The TrtMtcnn preprocessing code would take care of image resizing before feeding it into TensorRT engine for inference.

Nonetheless, I tested roughly the same code as the snippet you've shown.  It worked.  Please check out a clean copy of the code from this repo and try again.

jkjung-avt, 2020-08-12:
@jiagh2010  Do you have any update on this issue?

closed 2020-08-18 by jkjung-avt