jkjung-avt/tensorrt_demos #180: ERROR: coreReadArchive.cpp (38) - Serialization Error in verifyHeader: 0 (Version tag does not match)

adrianosantospb, 2020-08-05:
Hi, guys.

I'm trying to run the TensorRT model converted (YOLOV3-standard, ONNX, TensorRT) into the Jetson RX. I've created a service called serving_couting using Dockerfile. The image used to create a service was **nvcr.io/nvidia/l4t-pytorch:r32.4.2-pth1.5-py3**. 
I've tested the service on my computer and it works well, but it doesn't work on Jetson RX. 
The model was converted on my computer and the TensorRT version used to do it was 7.1.3-1+cuda11.0 (on my PC) and on the Jetson RX 7.1.0.16-1+cuda10.2. It should be compatible, isn't it?

The full error message is: 

`[2020-08-05 16:52:28 +0000] [42] [INFO] Booting worker with pid: 42
serving_counting       | [2020-08-05 16:52:38 +0000] [6] [CRITICAL] WORKER TIMEOUT (pid:42)
serving_counting       | [TensorRT] ERROR: coreReadArchive.cpp (38) - Serialization Error in verifyHeader: 0 (Version tag does not match)
serving_counting       | [TensorRT] ERROR: INVALID_STATE: std::exception
serving_counting       | [TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
serving_counting       | [2020-08-05 16:52:38 +0000] [42] [INFO] Worker exiting (pid: 42)
serving_counting       | Exception ignored in: <bound method TrtYOLO.__del__ of <yolo.TrtYOLO object at 0x7f88551940>>
serving_counting       | Traceback (most recent call last):
serving_counting       |   File "/home/smartcounting/detector/yolo.py", line 510, in __del__
serving_counting       |     del self.stream
serving_counting       | AttributeError: stream
serving_counting       | [2020-08-05 16:52:39 +0000] [51] [INFO] Booting worker with pid: 51`

Do I forget something? Tks.

jkjung-avt, 2020-08-06:
You cannot build a TensorRT engine on a x86_64 PC and then use that engine on the Jetson.  You should build the engine on your target platform.

adrianosantospb, 2020-08-10:
Great. I've done the conversion from Yolov3 to ONNX. But I've got a problem converting from ONNX to TRT. I'd like to know where could I get the "TensorRT "Python API" to aarch64?

jkjung-avt, 2020-08-10:
It should be available on your Jetson system after you install a recent version of JetPack on it.

You could verify the installation by, say, `$ pip3 show tensorrt`.

closed 2020-08-12 by jkjung-avt