jkjung-avt/tensorrt_demos #181: Difference between results on images and on video

AleBasso80, 2020-08-06:
Hello,
I'm running the mtcnn demo. If I feed the network with a video taken by my webcam the faces are detected practically only in the first two frames, instead, if I feed a single frame (even one not recognzed in the video) the detection works, but the same image within a loop of different images is not recognized. It's weird, but it seems to me that the detection in the first 1/2 frames is much more effective, then something happens.

What am I missing?

Thanks!

jkjung-avt, 2020-08-07:
Please provide sample code to reproduce this problem.  Thanks.

AleBasso80, 2020-08-07:
Hello,
I made two tests by using a video that afterwards I split in its image frames. 

In the first test I didn't make any change to your code (MTCNN demo). In the first go I run the video, in the following goes I separately run each frame. Whilst in the former case I got only a few detection (but always a success in the first 2/3 frames), in the latter I obtained the correct detections in all the images.

In the second test I modified the source code to be able to run several images in a loop.

def loop_and_detect_folder(folder, mtcnn, minsize, sorting=False):
    """Continuously capture images from camera and do face detection."""
    full_scrn = False
    fps = 0.0
    tic = time.time()
    for _, _, files in os.walk(folder):
      files.sort(reverse=sorting)
      for fl in files:
        if fl.endswith((".jpg")):
          if cv2.getWindowProperty(WINDOW_NAME, 0) < 0:
              break
          img = cv2.imread(folder + '/' + fl)
          if img is not None:
              dets, landmarks = mtcnn.detect(img, minsize=minsize)
              print('{} face(s) found'.format(len(dets)))
              img = show_faces(img, dets, landmarks)
              img = show_fps(img, fps)
              cv2.imshow(WINDOW_NAME, img)
              toc = time.time()
              curr_fps = 1.0 / (toc - tic)
              # calculate an exponentially decaying average of fps number
              fps = curr_fps if fps == 0.0 else (fps*0.95 + curr_fps*0.05)
              tic = toc
          key = cv2.waitKey(0)
          if key == 27:  # ESC key: quit program
              break
          elif key == ord('F') or key == ord('f'):  # Toggle fullscreen
              full_scrn = not full_scrn
              set_display(WINDOW_NAME, full_scrn)

The sorting variable is used to reverse the analysis sequence. I run the same images used before. Whilst in the first test all the images were correctly recognized, in this case the results depended on the order, but the same order always led to the same results. If you give me an e-mail address, if you wish I can send you the images that I used.

Thank you!

jkjung-avt, 2020-08-07:
Sure.  Please send the test images and the modified trt_mtcnn.py (full code) to me.  I will try to reproduce the problem.

AleBasso80, 2020-08-07:
I just had the chance to test the code on another PC and everything was fine, then I rebuilt the modules on mine and now it works like a charm. I have no idea about what happened. 
I apologize for having bothered you. Thanks a lot for your great work!

closed 2020-08-07 by AleBasso80