jkjung-avt/tensorrt_demos #182: yolo3 fp16

zhaogangthu, 2020-08-07:
hello,

I build a trt engine with config for dynamic shape
`builder.fp16_mode = True
 builder.strict_type_constraints = True
config.set_flag(trt.BuilderFlag.FP16)
engine = builder.build_engine(network,config)`

but when I allocate buffers with:
`size = trt.volume(engine.get_binding_shape(binding)) * engine.max_batch_size
dtype = trt.nptype(engine.get_binding_dtype(binding))
print('dtype:',dtype)
host_mem = cuda.pagelocked_empty(size, dtype)
device_mem = cuda.mem_alloc(host_mem.nbytes)
# Append the device buffer to device bindings.
bindings.append(int(device_mem))
 # Append to the appropriate list.
if engine.binding_is_input(binding):
    inputs.append(HostDeviceMem(host_mem, device_mem))
else:
     outputs.append(HostDeviceMem(host_mem, device_mem))`

I found the dtype of input and outputs is numpy.float32 which should be numpy.float16. Could you give me some suggestions, or could you share your code for allocate buffers？

Thanks.

jkjung-avt, 2020-08-07:
I think we need to feed "numpy.float32" data to TensorRT engines for inference, even when the engine is built with FP16 or INT8 flag.

zhaogangthu, 2020-08-07:
> I think we need to feed "numpy.float32" data to TensorRT engines for inference, even when the engine is built with FP16 or INT8 flag.

Yes，you are right. Thanks!

zhaogangthu, 2020-08-07:
> I think we need to feed "numpy.float32" data to TensorRT engines for inference, even when the engine is built with FP16 or INT8 flag.

hello,

I meet another problem. In my test, fp16 is not faster than fp32，is it normal? or if I convert the engine input to fp16 and feed fp16，would it be faster? 

Many thanks for your suggestions.

jkjung-avt, 2020-08-07:
It might be the case if your model contains many nodes/ops which do not support FP16.  (During inference, when TensorRT encounters borders between a FP16 node and a FP32 node, it needs to do FP16/FP32 conversion on the tensor.)

jkjung-avt, 2020-08-07:
One additional suggestion.  You could use "trtexec" to profile your TensorRT engine, in both FP16 and FP32 modes.  That way you could which nodes/ops are taking relative long to execute in the FP16 engine.

I've discussed how to do that in my [Applying TensorRT on My tf.keras ImageNet Models](https://jkjung-avt.github.io/trt-keras-imagenet/) blog post.

zhaogangthu, 2020-08-10:
> One additional suggestion. You could use "trtexec" to profile your TensorRT engine, in both FP16 and FP32 modes. That way you could which nodes/ops are taking relative long to execute in the FP16 [...]

Thanks. That's a really nice suggestion!

closed 2020-08-12 by jkjung-avt