jkjung-avt/tensorrt_demos #184: TypeError: buffer is too small for requested array

ashiqak, 2020-08-11:
nvidia@nvidia:~/yolo_test/tensorrt_demos/yolo$ **python3 yolo_to_onnx.py --model yolov3-custom-320x320 --category_num 1**
Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 957, in <module>
    main()
  File "yolo_to_onnx.py", line 940, in main
    verbose=True)
  File "yolo_to_onnx.py", line 451, in build_onnx_graph
    params)
  File "yolo_to_onnx.py", line 323, in load_conv_weights
    conv_params, 'conv', 'weights')
  File "yolo_to_onnx.py", line 351, in _create_param_tensors
    conv_params, param_category, suffix)
  File "yolo_to_onnx.py", line 382, in _load_one_param_type
    buffer=self.weights_file.read(param_size * 4))
**TypeError: buffer is too small for requested array**

can any one help
Thanks!

jkjung-avt, 2020-08-11:
This error message generally means that the weights file contains too few values for loading up the model (as defined in the cfg file).

Have you verified that "darknet" could run your custom trained model successfully?

ashiqak, 2020-08-12:
yeah!, I can able to run the model in local system successfully only in nano i facing this issue.
don't know were went wrong.
this model is trained in 320x320 dimention but 416x416 does'nt shown up any error it run's quietly
**Model**:yoloV3
can any one help.
Thanks!

jkjung-avt, 2020-08-31:
@ashiqak  Are you still having problem with this?  I'm not able to offer any help since I cannot reproduce the problem.  I will close this issue if there's no further update.

closed 2020-09-03 by jkjung-avt