jkjung-avt/tensorrt_demos #189: Upgrading instructable

krambriw, 2020-08-11:
Hi Jung,
This is a great software & guide you have provided. Working excellent  on my Jetson Nanos
I just finished the demos you provided, in particular #5 YOLO v4, building the engines etc. Now I just saw the demo #6 and of course got interested to optimize! But tell me, do I have to start all over again? Clone again, building engines etc etc?
Or is there a "smart" way how I can just copy missing files from the git to my installation? I clearly see that the folder plugin is missing but whats more? Is it possible to fix this so I can re-use what is already done?

Best regards, Walter

krambriw, 2020-08-11:
I decided to redo anyway /would be nice to know if it could be done easier as asked previous)

At the step 
$ cd ${HOME}/project/tensorrt_demos/plugins
$ make

it fails with the following error messages below. Any hint is appreciated
Best regards, Walter

```
wk@jetson:~/project/tensorrt_demos/plugins$ make
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_53,code=sm_53 -gencode arch=compute_53,code=compute_53 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(38): error: not a class or struct name

yolo_layer.h(44): error: member function declared with "override" does not override a base class member

yolo_layer.h(46): error: member function declared with "override" does not override a base class member

yolo_layer.h(51): error: member function declared with "override" does not override a base class member

yolo_layer.h(53): error: member function declared with "override" does not override a base class member

yolo_layer.h(55): error: member function declared with "override" does not override a base class member

yolo_layer.h(57): error: member function declared with "override" does not override a base class member

yolo_layer.h(59): error: member function declared with "override" does not override a base class member

yolo_layer.h(61): error: member function declared with "override" does not override a base class member

yolo_layer.h(63): error: member function declared with "override" does not override a base class member

yolo_layer.h(65): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(65): error: member function declared with "override" does not override a base class member

yolo_layer.h(69): error: member function declared with "override" does not override a base class member

yolo_layer.h(71): error: member function declared with "override" does not override a base class member

yolo_layer.h(73): error: member function declared with "override" does not override a base class member

yolo_layer.h(75): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(75): error: member function declared with "override" does not override a base class member

yolo_layer.h(77): error: member function declared with "override" does not override a base class member

yolo_layer.h(79): error: member function declared with "override" does not override a base class member

yolo_layer.h(81): error: member function declared with "override" does not override a base class member

yolo_layer.h(83): error: member function declared with "override" does not override a base class member

yolo_layer.h(85): error: member function declared with "override" does not override a base class member

yolo_layer.h(87): error: member function declared with "override" does not override a base class member

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: member function declared with "override" does not override a base class member

yolo_layer.h(89): error: expected a ";"

yolo_layer.h(91): error: member function declared with "override" does not override a base class member

yolo_layer.h(66): error: name followed by "::" must be a class or namespace name

yolo_layer.h(119): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(119): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::createPlugin"

yolo_layer.h(119): error: member function declared with "override" does not override a base class member

yolo_layer.h(121): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(121): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::deserializePlugin"

yolo_layer.h(121): error: member function declared with "override" does not override a base class member

yolo_layer.cu(142): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(142): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(172): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(267): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(325): error: identifier "IPluginV2IOExt" is undefined

/usr/include/aarch64-linux-gnu/NvInfer.h(5935): error: object of abstract class type "nvinfer1::YoloPluginCreator" is not allowed:
            pure virtual function "nvinfer1::IPluginCreator::createPlugin" has no overrider
            pure virtual function "nvinfer1::IPluginCreator::deserializePlugin" has no overrider
          detected during instantiation of class "nvinfer1::PluginRegistrar<T> [with T=nvinfer1::YoloPluginCreator]"
yolo_layer.cu(334): here

41 errors detected in the compilation of "/tmp/tmpxft_00004428_00000000-6_yolo_layer.cpp1.ii".
Makefile:46: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1
wk@jetson:~/project/tensorrt_demos/plugins$

```

jkjung-avt, 2020-08-12:
This looks like incompatibility of API between different versions of TensorRT.  As mentioned in the README.md, my current Demo #6 code has only be tested against TensorRT 7.  What version of TensorRT are you using?

krambriw, 2020-08-12:
Right, yes, I have 5.1.6 only. Should I upgrade to JetPack-4.4?

And what with my first setup of tensorrt_demos, could that be upgraded by simply copying some files from the repo?

jkjung-avt, 2020-08-12:
I just did some quick test.  TensorRT 6 should also work.

Anyway, if you upgrade to JetPack-4.4 (TensorRT 7), the current code in this repo should just work.  For now, TensorRT engines built by version 5 could not be used by version 7.  You should build the TensorRT engines all over again.

krambriw, 2020-08-12:
Hi again, tried to upgrade using NVIDIA SDK Manager but the Jetson Nano is not recognized. It seems it is connected to the ubuntu host but still not recognized. Do you have any hint? Something I need to do on the Jetson?

[image]

jkjung-avt, 2020-08-12:
It's easier to just follow NVIDIA's [Getting Started With Jetson Nano Developer Kit](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit) guide to set up Jetson Nano with the latest JetPack (4.4).

krambriw, 2020-08-12:
OK, I will start with a fresh image
(Anyway, the problem was I forgot the jumper to force recovery, after this the sdk manager found the Nano but there was nothing to update anyway, guess I have too old installations)

closed 2020-08-12 by krambriw