jkjung-avt/tensorrt_demos #191: Questions with MTCNN pipeline

pushkalkatara, 2020-08-12:
1. I noticed that PRelu is converted to Relu to support TensorRT Caffe Parser. Would it affect accuracy in any sense?

2. Can we create one TRT Engine for all three architectures (P-O-R Net) or we can achieve more optimisation if we have 3 different engines? 

3. If we go by the approach Pytorch model -> ONNX -> TensorRT engine, ONNX and TensorRT7 now support PRelu, would that be better in terms of accuracy?

jkjung-avt, 2020-08-12:
1. I think the accuracy is not affected.  The 2 ReLU replacement is mathematically identical to the original PReLU.

2. The operations in-between PNet and RNet and in-between RNet and ONet could not easily be implemented by TensorRT plugins.  I think this proposal is not feasible.

3. No effect on accuracy, but might improve inference speed.

closed 2020-08-18 by jkjung-avt

pushkalkatara, 2020-08-18:
Thanks :D