jkjung-avt/tensorrt_demos #192: Problem with img = self.cam.read() in trt_ssd_async

PatrikPatrik2, 2020-08-12:
I use the trt_ssd_async with a rtsp feed, and it usually work perfectly for many hours, but thereafter my system freezes. From the error message it seems to be related to the  img = self.cam.read(). I modified the code slightly:

`        while self.running:
            img = self.cam.read()
            if img is None:
                print("img returned None.")
                break
            else:`

when the error occurs i get this:

`[ WARN:2] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (488) isPipelinePlaying OpenCV | GStreamer warning: unable to query pipeline state
[ WARN:2] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module rtspsrc0 reported: Could not read from resource.
WARNING:root:grab_img(): cap.read() returns None...
img returned None.
TrtThread: stopped...`

and the program stops.
Is there some way to modify the code such that it does not stop if self.cam.read() return None?
If I remove the "break" it will just print "img returned None." forever

jkjung-avt, 2020-08-13:
RTSP is usually not a reliable video source.  The streaming could break due to a variety of reasons, such as network packet loss, network connection reset, or malfunctioning of the camera.

In my experience, when cv2.VideoCapture's read() function returns no img, it's likely that one of the errors mentioned-above has occurred.  You'll have to close the VideoCapture object and try to re-open it (maybe retry many times), in order to recover from that.

My code is for demo purposes only.  I don't intend to handle those errors in the demo code.

PatrikPatrik2, 2020-08-13:
Thanks.

I tried this:
`while self.running:
            img = self.cam.read()
            if img is None:
                while img is None:
                    print("img returned None.")
                    print("Close Cam")
                    cam.stop()
                    print("Release Cam")
                    cam.release()
                    print("Start Cam")  
                    cam.start()  # let camera start grabbing frames
                    print("All done")
                    img = self.cam.read() 
                break`

but it got stuck in the while loop - self.cam.read() gave None all the time - what am I missing?

jkjung-avt, 2020-08-13:
It would be easier to handle this in the main program.  Just close the camera object and re-instantiate a new one (retry a few times if needed).

PatrikPatrik2, 2020-08-13:
Sorry for beeing a noob, how can I get the information that self.cam.read() in def run(self) returned None to the def main(): such that the main function can close and re-instantiate

jkjung-avt, 2020-08-13:
`img = self.cam.read()` would be None at this line.

https://github.com/jkjung-avt/tensorrt_demos/blob/8a93c9298a68f976eb4745851847b963f2d9c452/trt_ssd_async.py#L100

When that happens, you know you need to close camera and reinstiate another one.  It might require some redesign to get this working on the async version of the code (trt_ssd_async.py)...  It'd be easier to handle this in the sync version of the code (trt_ssd.py).

PatrikPatrik2, 2020-08-13:
Just double cheking that I have understood:
I will use trt_ssd.py
when self.cam.read() = None, I will start a new instance in the main program by:
`    cam.stop()
    cam.release()
    cv2.destroyAllWindows()
    cam = Camera(args)
    cam.open()
    if not cam.is_opened:
        sys.exit('Failed to open camera!')

    cls_dict = get_cls_dict(args.model.split('_')[-1])
    trt_ssd = TrtSSD(args.model, INPUT_HW)

    cam.start()
    open_window(WINDOW_NAME, args.image_width, args.image_height,
                'Camera TensorRT SSD Demo for Jetson Nano')
    vis = BBoxVisualization(cls_dict)
    loop_and_detect(cam, trt_ssd, conf_th=0.3, vis=vis)
`  

Have I understood correctly?

jkjung-avt, 2020-08-13:
It would be easier if you just instantiate "cam" in the TrtThread.  For example, save "args" in __init__() and modify run() as below.  Also remember to remove "cam" related code in main().

```
    def run(self):
        """Run until 'running' flag is set to False by main thread."""
        global s_img, s_boxes, s_confs, s_clss

        cam = None

        print('TrtThread: loading the TRT SSD engine...')
        self.trt_ssd = TrtSSD(self.model, INPUT_HW)
        print('TrtThread: start running...')
        self.running = True
        while self.running:
            if not cam:
                cam = Camera(args)
                cam.open()
                if not cam.is_opened:
                    # retry
                cam.start()
            img = self.cam.read()
            if img:
                boxes, confs, clss = self.trt_ssd.detect(img, self.conf_th)
                with self.condition:
                    s_img, s_boxes, s_confs, s_clss = img, boxes, confs, clss
                    self.condition.notify()
            else:
                cam.stop()
                cam.release()
                cam = None
        cam.stop()
        cam.release()
        del self.trt_ssd
        print('TrtThread: stopped...')
```

PatrikPatrik2, 2020-08-13:
I tried to remove cam from main and send args to init() and used the modified run():

`
def __init__(self, condition, args, conf_th):
        """__init__

        # Arguments
            condition: the conditio:n variable used to notify main
                       thread about new frame and detection result
            cam: the camera object for reading input image frames
            model: a string, specifying the TRT SSD model
            conf_th: confidence threshold for detection
        """
        threading.Thread.__init__(self)
        self.condition = condition
        self.model = args.model
        self.conf_th = conf_th
        self.cuda_ctx = None  # to be created when run
        self.trt_ssd = None   # to be created when run
        self.running = False

    def run(self):
        """Run until 'running' flag is set to False by main thread."""
        global s_img, s_boxes, s_confs, s_clss

        cam = None

        print('TrtThread: loading the TRT SSD engine...')
        self.trt_ssd = TrtSSD(self.model, INPUT_HW)
        print('TrtThread: start running...')
        self.running = True
        while self.running:
            if not cam:
                #cam = Camera(self.args)
                cam = Camera(args)
                cam.open()
                if not cam.is_opened:
                    # retry
                    cam.start()
                img = self.cam.read()
            if img:
                boxes, confs, clss = self.trt_ssd.detect(img, self.conf_th)
                with self.condition:
                    s_img, s_boxes, s_confs, s_clss = img, boxes, confs, clss
                    self.condition.notify()
            else:
                cam.stop()
                cam.release()
                cam = None
        cam.stop()
        cam.release()
        del self.trt_ssd
        print('TrtThread: stopped...')`

`
    def main():
    args = parse_args()

    cls_dict = get_cls_dict(args.model.split('_')[-1])

    cuda.init()  # init pycuda driver

    open_window(WINDOW_NAME, args.image_width, args.image_height,
                'Camera TensorRT SSD Demo for Jetson Nano')
    vis = BBoxVisualization(cls_dict)
    condition = threading.Condition()
    trt_thread = TrtThread(condition, args, conf_th=0.3)
    trt_thread.start()  # start the child thread
    loop_and_display(condition, vis)
    trt_thread.stop()   # stop the child thread

    cam.stop()
    cam.release()
    cv2.destroyAllWindows()

`

But this resulted in error:

` 
2020-08-13 22:03:47.861506: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
TrtThread: loading the TRT SSD engine...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "trt_ssd_async.py", line 91, in run
    self.trt_ssd = TrtSSD(self.model, INPUT_HW)
  File "/media/pi/SSD1/Orginal/utils/ssd.py", line 88, in __init__
    self.stream = cuda.Stream()
pycuda._driver.LogicError: explicit_context_dependent failed: invalid device context - no currently active context?

Exception ignored in: <bound method TrtSSD.__del__ of <utils.ssd.TrtSSD object at 0x7f2fcefac8>>
Traceback (most recent call last):
  File "/media/pi/SSD1/Orginal/utils/ssd.py", line 93, in __del__
    del self.stream
AttributeError: stream
`

What have I done wrong?

jkjung-avt, 2020-08-14:
> File "/media/pi/SSD1/Orginal/utils/ssd.py", line 88, in init
> self.stream = cuda.Stream()
> pycuda._driver.LogicError: explicit_context_dependent failed: invalid device context - no currently active context?

This error is related to cuda context in the child thread.

I suspect you are not using my latest code in the repo.  Please pull the latest code and try again.  I recently moved the `cuda.Device(0).make_context()` code from here:

https://github.com/jkjung-avt/tensorrt_demos/blob/8a93c9298a68f976eb4745851847b963f2d9c452/trt_ssd_async.py#L95

to here:

https://github.com/jkjung-avt/tensorrt_demos/blob/8a93c9298a68f976eb4745851847b963f2d9c452/utils/ssd.py#L78

PatrikPatrik2, 2020-08-14:
You were right, apparently I used an old version. I used the new version and made alteration. The program run a bit longer, but I got this error:
`
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Allocating new output: 1920x1088 (x 11), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1920, nFrameHeight = 1080 
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (896) open OpenCV | GStreamer warning: unable to query duration of stream
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "trt_ssd_async_August_1.py", line 105, in run
    img = self.cam.read()
AttributeError: 'TrtThread' object has no attribute 'cam'

`
The modified code is below (sorry for the bad formating).
I removed cam from TrtThread and used args instead

TrtThread(condition, args, conf_th=0.3)

What shall I do to fix the "AttributeError: 'TrtThread' object has no attribute 'cam'" error? Everything I have tried has failed

`
import sys
import time
import argparse
import threading

import cv2
import pycuda.driver as cuda

from utils.ssd_classes import get_cls_dict
from utils.ssd import TrtSSD
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

WINDOW_NAME = 'TrtSsdDemoAsync'
INPUT_HW = (300, 300)
SUPPORTED_MODELS = [
    'ssd_mobilenet_v1_coco',
    'ssd_mobilenet_v1_egohands',
    'ssd_mobilenet_v2_coco',
    'ssd_mobilenet_v2_egohands',
]

# These global variables are 'shared' between the main and child
# threads.  The child thread writes new frame and detection result
# into these variables, while the main thread reads from them.
s_img, s_boxes, s_confs, s_clss = None, None, None, None

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time object detection with TensorRT optimized '
            'SSD model on Jetson Nano')
    parser = argparse.ArgumentParser(description=desc)
    parser = add_camera_args(parser)
    parser.add_argument('--model', type=str, default='ssd_mobilenet_v2_coco',
                        choices=SUPPORTED_MODELS)
    args = parser.parse_args()
    return args

class TrtThread(threading.Thread):
    """TrtThread

    This implements the child thread which continues to read images
[... 102 lines cut ...]
            set_display(WINDOW_NAME, full_scrn)

def main():
    args = parse_args()

    cls_dict = get_cls_dict(args.model.split('_')[-1])

    cuda.init()  # init pycuda driver

    open_window(WINDOW_NAME, args.image_width, args.image_height,
                'Camera TensorRT SSD Demo for Jetson Nano')
    vis = BBoxVisualization(cls_dict)
    condition = threading.Condition()
    trt_thread = TrtThread(condition, args, conf_th=0.3)
    trt_thread.start()  # start the child thread
    loop_and_display(condition, vis)
    trt_thread.stop()   # stop the child thread

    cam.stop()
    cam.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
`

jkjung-avt, 2020-08-15:
You should have replaced this line:

`                img = self.cam.read()`

with this:

`                img = cam.read()`

PatrikPatrik2, 2020-08-16:
When I changed the line I got this strange error:

`
(TrtSsdDemoAsync:30404): GStreamer-CRITICAL **: 21:29:20.511: gst_mini_object_unref: assertion 'mini_object != NULL' failed
NvMMLiteOpen : Block : BlockType = 261 

NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Allocating new output: 1920x1088 (x 11), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1920, nFrameHeight = 1080 
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (896) open OpenCV | GStreamer warning: unable to query duration of stream
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "trt_ssd_async_August_1.py", line 112, in run
    cam.stop()
  File "/media/pi/SSD1/Orginal/utils/camera.py", line 213, in stop
    self.thread.join()
AttributeError: 'NoneType' object has no attribute 'join'
`

jkjung-avt, 2020-08-17:
Sorry, I'm not sure what the problem is.  As stated earlier, I don't plan to spend more time on this issue.  I think you could use my TrtSSD code and redesign the RTSP video input code by yourself.  Good luck.

PatrikPatrik2, 2020-08-17:
OK, thanks for your effort!

closed 2020-08-18 by jkjung-avt