jkjung-avt/tensorrt_demos #193: Walking the demo #6 optimizing with TensorRT plugin

krambriw, 2020-08-13:
Hi, started yesterday with a fresh dowloaded image for the Jetson Nano

Everything went smoothly until trying to install the "onnx==1.4.1". After extensive search I found that this problem had already been identified and mentioned as an issue in here

This should be stated in the instructions that running that script "install_protobuf-3.8.0.sh" takes several hours, at least it took that for me. After running the script, the onnx installed correctly

When running YOLOv4 yolov4-tiny-416 or yolov4-416 (which is what I focus on) I get an error message regarding a canberra-gtk-module but it works anyway. Is this something I need to add to sort this out?
Gtk-Message: 06:19:40.952: Failed to load module "canberra-gtk-module"

After following the instructions in #6 optimizing and testing I just get this error message for both yolov4-tiny-opt-416 and yolov4-opt-416  

```
wk@JN228:~/trt_projects/tensorrt_demos$ python3 trt_yolo.py --model yolov4-opt-416 --image --filename ${HOME}/Pictures/dog.jpg
[TensorRT] ERROR: INVALID_ARGUMENT: getPluginCreator could not find plugin YoloLayer_TRT version 1
[TensorRT] ERROR: safeDeserializationUtils.cpp (323) - Serialization Error in load: 0 (Cannot deserialize plugin since corresponding IPluginCreator not found in Plugin Registry)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
  File "/home/wk/trt_projects/tensorrt_demos/utils/yolo.py", line 506, in __init__
    self.context = self._create_context()
  File "/home/wk/trt_projects/tensorrt_demos/utils/yolo.py", line 408, in _create_context
    return self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "trt_yolo.py", line 122, in <module>
    main()
  File "trt_yolo.py", line 108, in main
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
  File "/home/wk/trt_projects/tensorrt_demos/utils/yolo.py", line 512, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo.TrtYOLO object at 0x7f8b5108d0>>
Traceback (most recent call last):
  File "/home/wk/trt_projects/tensorrt_demos/utils/yolo.py", line 516, in __del__
    del self.outputs
AttributeError: outputs

```

jkjung-avt, 2020-08-13:
1. Be sure to do `make` in the "plugins/" subdirectory.
2. Be sure to add "--with_plugins" (or "-p") when you do "onnx_to_tensorrt.py".
3. Be sure to also add "--with_plugins" when you run "trt_yolo.py".

jkjung-avt, 2020-08-13:
> This should be stated in the instructions that running that script "install_protobuf-3.8.0.sh" takes several hours, at least it took that for me. After running the script, the onnx installed correctly.

Thanks for pointing this out.  I shall find time to improve the documentation.  The "install_protobuf-3.8.0.sh" script would take a couple of hours, indeed.

> Gtk-Message: 06:19:40.952: Failed to load module "canberra-gtk-module"

This could be ignored.  Alternatively, you could do the following to get rid of that warning.  I mentioned it in my [JetPack-4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/) blog post.

```
$ sudo apt-get install -y libcanberra-gtk-module libcanberra-gtk3-module
```

krambriw, 2020-08-13:
Oops, my fault, missed the --with_plugins, sorry
Works perfect now!!!
Only the canberra error left now
Thanks a lot for great sw & support! Really impressive,
Best regards, Walter

krambriw, 2020-08-13:
Canberry error fixed after installation
Closing this issue now
Thanks again!

closed 2020-08-13 by krambriw