jkjung-avt/tensorrt_demos #196: How to increase the display window size when running YOLOv4 tensorrt version on USB webcam?

Farjad3253, 2020-08-18:
Hello! I want to ask a question in which I am stuck and not able to find a solution. The issue is that when I run this command :
python3 trt_yolo.py --model yolov3-custom-416 --category_num 1 --usb 
Then when the camera feed is displayed in the window then that window size is very small. I want to increase that window size. This is the last thing in which I am stuck. So kindly tell me the solution to this issue.

jkjung-avt, 2020-08-18:
You could specify "--width" and "--height" on the command line.  For example,

```
python3 trt_yolo.py --model yolov3-custom-416 --category_num 1 --usb --width 1280 --height 720
```

Farjad3253, 2020-08-18:
@jkjung-avt . Thanks a lot Sir. You always helped me and responded very fast. Sir I will run the program again and if found any issue then will let you know. Regards!

jkjung-avt, 2020-08-18:
I just updated my "Camera" class code in this repo.  The new code runs faster for certain image/video inputs, and the command-line arguments are modified/simplified.  If you check out the latest code, be sure to refer to the updated README.md.

Farjad3253, 2020-08-18:
> The new code runs faster for certain image/video inputs

@jkjung-avt . Do you mean that on usb webcam it will give higher FPS? Like for example currently I am getting 4.5 FPS on my custom yolov4 of size= 416x416. Now what I need is to again re-clone your repo and check for the results on demo 5?
Do I need to change some command line arguments or not?

jkjung-avt, 2020-08-18:
Just `git pull` the latest code.  You don't need to re-build the TensorRT engine (since only camera related code is updated).

Command-line arguments have been modified (simplified) a little bit.  Refer to README.md or https://github.com/jkjung-avt/tensorrt_demos/issues/196#issuecomment-675221450.

closed 2020-08-18 by jkjung-avt

Farjad3253, 2020-08-18:
@jkjung-avt . Ok thanks a lot!