jkjung-avt/tensorrt_demos #197: Failed to open camera for YoloV3/Yolov4

XisHungry, 2020-08-18:
Hi, great tutorial. The instructions were clear and detailed.
I managed to run a detection for dog.jpg and got its outcome. I wanted to perform a detection using a webcam however i encountered this error: ERROR: failed to open camera! I can ensure that the camera is working as i was using it for detection via Darknet.
This is the code i ran:
```
python3 trt_yolo.py --model yolov3-tiny-288 --usb --width 1280 --height 720
```
May i know where did i go wrong? Perhaps i missed out on something in the instructions.
Once again thank you for the guide!

jkjung-avt, 2020-08-18:
I just modify the command-line arguments for camera a couple of hours ago.  Please pull the latest code from the repo.  And note that you need to specify V4L2 device number (/dev/video?)  when you use `--usb` (append a "0" to this command-line argument):

```
python3 trt_yolo.py --model yolov3-tiny-288 --usb 0 --width 1280 --height 720
```

If it still doesn't work, please make sure your USB webcam supports 1280x720 resolution.  You could try, for example, 640x480 instead.

XisHungry, 2020-08-18:
Oh, thanks! I will try it. Thank you for the prompt reply!

XisHungry, 2020-08-19:
Hi, i realized i was using OpenCV 3.4.1 instead. This was my error, after converting to OpenCV 3.4.6. The entire project was working!

I would like to thank you for this amazing guide and assistance provided. Thank you!

closed 2020-08-19 by XisHungry

jkjung-avt, 2020-08-19:
Glad to hear that.