jkjung-avt/tensorrt_demos #198: yolo demo5 plugin update

IbrahimBond, 2020-08-18:
in your last update of yolo demos, there is noway to change the anchors. how is it possible to do so? 

i have a custom model and the resulting bboxes have the wrong dimensions and i doubt this is the reason.

can you confirm this?

jkjung-avt, 2020-08-18:
With the latest code, you don't need to modify anchor values in python code anymore.  My onnx_to_tensorrt.py would parse the cfg file and set anchor values accordingly when creating the "yolo_layer" plugins.  So you just need to make sure you have the correct anchor values in your cfg file.  For example, the following is from the official [yolov4.cfg](https://raw.githubusercontent.com/AlexeyAB/darknet/master/cfg/yolov4.cfg):

```
......
[yolo]
mask = 0,1,2
anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
......
```

If you'd like to check out the relevant source code, look [here](https://github.com/jkjung-avt/tensorrt_demos/blob/4b40441d838c1e44b1e355f50729590d0ca18c52/yolo/plugins.py#L61) and [here](https://github.com/jkjung-avt/tensorrt_demos/blob/4b40441d838c1e44b1e355f50729590d0ca18c52/yolo/plugins.py#L123).

IbrahimBond, 2020-08-19:
i figured..

for some reason any detected object is giving me the same bbox as follows:

415 2 415 2
415 2 415 2
415 2 415 2
415 2 415 2
415 2 415 2
415 2 415 2 

which is basically like plotting a single point.

jkjung-avt, 2020-08-19:
Did you set "width" and "height" correctly in the cfg file?

```
[net]
......
width=608
height=608
......
```

IbrahimBond, 2020-08-19:
yes. i used the same weights and cfg i used before to re produce the trt file

jkjung-avt, 2020-08-19:
Could you provide your cfg file?  I will take a look.

IbrahimBond, 2020-08-19:
sure.

[net]
# Testing
batch=64
subdivisions=4
# Training
# batch=64
# subdivisions=2
width=416
height=416
channels=3
momentum=0.9
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.001
burn_in=1000
max_batches = 500200
policy=steps
steps=400000,450000
scales=.1,.1

[convolutional]
batch_normalize=1
filters=16
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=64
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=128
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=2

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

[maxpool]
size=2
stride=1

[convolutional]
batch_normalize=1
filters=1024
size=3
stride=1
pad=1
activation=leaky

###########

[convolutional]
batch_normalize=1
filters=256
size=1
stride=1
pad=1
activation=leaky

[convolutional]
batch_normalize=1
filters=512
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=27
activation=linear

[yolo]
mask = 3,4,5
anchors =  45, 66,  65,110,  88,156, 103,213, 124,292, 162,372
classes=4
num=6
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

[route]
layers = -4

[convolutional]
batch_normalize=1
filters=128
size=1
stride=1
pad=1
activation=leaky

[upsample]
stride=2

[route]
layers = -1, 8

[convolutional]
batch_normalize=1
filters=256
size=3
stride=1
pad=1
activation=leaky

[convolutional]
size=1
stride=1
pad=1
filters=27
activation=linear

[yolo]
mask = 0,1,2
anchors =  45, 66,  65,110,  88,156, 103,213, 124,292, 162,372
classes=4
num=6
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1

IbrahimBond, 2020-08-19:
i have found out the source of the problem, but it doesnt make sense for me. i have the preprocessing done with frame reading in a separate thread. so when i call detect() i send a frame that is already preprocessed using the same way _postprocess_yolo() does it. and i also send  the original WH for the _postprocess_yolo() so there should be no problem. 

sending the frame as is to the detect() and preprocessing it there works fine. 

do you have any idea why would this occur?

thanks

jkjung-avt, 2020-08-19:
If you scale the image to 416x416 before calling TrtYOLO.detect(), the resulting bounding box coordinates would be for the 416x416 image (not coordinates on the original image).

Otherwise, I don't know why you see the following problem...

> for some reason any detected object is giving me the same bbox as follows:
> 
> 415 2 415 2
> 415 2 415 2
> 415 2 415 2
> 415 2 415 2
> 415 2 415 2
> 415 2 415 2
> 
> which is basically like plotting a single point.

IbrahimBond, 2020-08-24:
this is the same as your detect function, but with small modification:

def detect(self, img, shape_orig_WH, conf_th=0.3):
        """Detect objects in the input image."""
        #img_resized = _preprocess_yolo(img, self.input_shape)
    
        # Set host input to the image. The do_inference() function
        # will copy the input to the GPU before executing.
        self.inputs[0].host = np.ascontiguousarray(img_resized)
        if self.cuda_ctx:
            self.cuda_ctx.push()
        trt_outputs = self.inference_fn(
            context=self.context,
            bindings=self.bindings,
            inputs=self.inputs,
            outputs=self.outputs,
            stream=self.stream)
        if self.cuda_ctx:
            self.cuda_ctx.pop()

        boxes, scores, classes = _postprocess_yolo(
            trt_outputs,shape_orig_WH[0], shape_orig_WH[1], conf_th)

        # clip x1, y1, x2, y2 within original image
        boxes[:, [0, 2]] = np.clip(boxes[:, [0, 2]], 0, img.shape[1]-1)
        boxes[:, [1, 3]] = np.clip(boxes[:, [1, 3]], 0, img.shape[0]-1)
        return boxes, scores, classes

anyway i'll go ahead and close the issue

closed 2020-08-24 by IbrahimBond