jkjung-avt/tensorrt_demos #205: Whether to support yolov4-tiny,and outputing of the target box

Shame-fight, 2020-08-25:
Thanks for your work, I noticed that the final detection result does not seem to output the coordinates of the target frame (like x, y, w, h), is my understanding correct? Can you add the function of outputting the coordinate boxing to the code?

jkjung-avt, 2020-08-25:
> Whether to support yolov4-tiny

Yes.  Please refer to [Demo #5: YOLOv4](https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4) for details.

> I noticed that the final detection result does not seem to output the coordinates of the target frame (like x, y, w, h), is my understanding correct? Can you add the function of outputting the coordinate boxing to the code?

Please refer to [utils/yolo_with_plugins.py](https://github.com/jkjung-avt/tensorrt_demos/blob/a5e32acce80468a9cb1e8ba58bc82c72a6bbf5c6/utils/yolo_with_plugins.py#L83).  The TensorRT yolov4/yolov3 engine would output x, y, w, h (in the range of 0.0~1.0).  And the _postprocess_yolo() function would convert them into x1, y1, x2, y2 (or xmin, ymin, xmax, ymax) in "pixels".  Also refer to [this line of code](https://github.com/jkjung-avt/tensorrt_demos/blob/a5e32acce80468a9cb1e8ba58bc82c72a6bbf5c6/utils/yolo_with_plugins.py#L127).

```
        boxes = np.concatenate([xx, yy, xx+ww, yy+hh], axis=1) + 0.5
        boxes = boxes.astype(np.int)
```

Shame-fight, 2020-08-25:
cool!Thank you for such a quick response, I will try your code.

closed 2020-08-25 by Shame-fight