jkjung-avt/tensorrt_demos #206: A suggestion and need about trt_yolov4 on ros

Shame-fight, 2020-08-25:
Can this function be compiled into ros nodes, such as trt-yolov4. For robots, ros is essential and convenient.If you can integrate all the functions (googlenet/sdd/mtcnn) into the ros package at the same time, it will be perfect.

Shame-fight, 2020-08-26:
I found a similar [project,](https://github.com/kumuji/trt_yolo_ros) but the author no longer updates yolov4 and yolov4-tiny, so stop at yolov3, do you have an idea to update yolov4 to ros package?

jkjung-avt, 2020-08-27:
Thanks for the suggestion.  I'm not familiar with ROS at this moment.  I will take a look when I have a chance.

By the way, I think it shouldn't be too difficult to integrate my TensorRT YOLO code to other applications.  You only need to make sure the requirements (TensorRT library, cv2, pycuda) are installed properly.  And copy over these 3 files: "yolo/yolov4-xxx.trt",  "plugins/libyolo_layer.so", and "utils/yolo_with_plugins.py".  Then you can do inference (detect objects) using the "TrtYOLO" class.

closed 2020-09-15 by jkjung-avt