jkjung-avt/tensorrt_demos #207: How to raise an alarm from a GPIO pin of Jetson Nano once an object is detected?

Farjad3253, 2020-08-27:
Hello! I am in an issue that I want an external alarm that is connected with a digital pin of Jetson Nano to turn on once any object is detected. I want urgent help in this regard as my degree is at stake and this is the must requirement. I am using custom YOLOv4-416 trained on one class. Waiting for help.

jkjung-avt, 2020-08-27:
I have written a blog post about how to access GPIO as a non-root user on Jetson TX1: [Accessing TX1 GPIO as a Non-root User (ubuntu)](https://jkjung-avt.github.io/gpio-non-root/).

I think most of it should apply to Jetson Nano.  (But I have not tried it on Jetson Nano, though.  You'll have to solve issues by yourself.)

Farjad3253, 2020-08-27:
@jkjung-avt . What are your thoughts that if there is any addition that can be made in trt_yolo.py i.e. specifically in loop_and_detect() function for this issue?

Farjad3253, 2020-08-27:
@jkjung-avt I have also seen your post : Accessing TX1 GPIO as a Non-root User (ubuntu). But there is no info regarding once an object is detected then how to turn on an alarm using a digital pin of nano?

jkjung-avt, 2020-08-27:
You have to implement that by yourself...

Farjad3253, 2020-08-27:
@jkjung-avt . This is last thing which remains. Kindly give some hint so that I can move forward.

jkjung-avt, 2020-08-27:
I already did...

Farjad3253, 2020-08-27:
@jkjung-avt . I have seen that post but I am not getting it?

jkjung-avt, 2020-08-27:
I'm not able to help further.

closed 2020-08-27 by jkjung-avt

Farjad3253, 2020-08-27:
Ok Thanks a lot!