jkjung-avt/tensorrt_demos #208: conver yolo to onnx stop at building onnx gragh

momandai, 2020-08-27:
i run
```
python yolo_to_onnx.py -m yolov4-416
```
and it stop at building onnx graph, and memory is increasing
[image]

jkjung-avt, 2020-08-27:
I have never encountered such a problem.  Are you running this on a Jetson Nano?  Are you converting the pre-trained (downloaded) yolov4 model?

momandai, 2020-08-28:
> I have never encountered such a problem. Are you running this on a Jetson Nano? Are you converting the pre-trained (downloaded) yolov4 model?

i ran this on Jetson TX2, jetpack4.3
i installed protobuf3.8.0 https://github.com/jkjung-avt/jetson_nano/blob/master/install_protobuf-3.8.0.sh
installed   onnx 1.4.1
ran
```
./download_yolo.sh
```
but when i convert pre-trained model
```
python3 yolo_to_onnx.py -m yolov4-416
```
it stucked at building onnx graph

momandai, 2020-08-28:
it took about 8 minute to finish building onnx model, and   used 30% memory while converting to onnx, is it normal?
[image]

jkjung-avt, 2020-08-28:
As long as the program does not throw an error or exception, I think it's OK.  I have tested my tensorrt yolo code on both Jetson Nano and Xavier NX, but not TX2 yet.  I think it'd work anyway.

momandai, 2020-08-28:
thx, i tested yolov3-416.trt on TX2,  about 5.5 FPS

closed 2020-08-28 by momandai