jkjung-avt/tensorrt_demos #209: Suggestion rtsp output

tomcatxx, 2020-08-30:
Hi, first thanks for all the nice examples. Helped me a lot.
Im playing around with object detection in front of an cctv system to trigger alerts.
Would it be possible to get an rtsp stream output? 
I would use deepstream but wasnt able to get yolo4 running.

jkjung-avt, 2020-08-31:
Thanks for the suggestion.  I might look into that when I have time, but no promise.

tomcatxx, 2020-09-03:
nice ty 👍

jkjung-avt, 2020-09-14:
I created a MJPEG sample.  You could run it and view detection results remotely using a web browser.  Please check https://github.com/jkjung-avt/tensorrt_demos/issues/226#issuecomment-691689456.

I don't plan to develop RTSP streaming code since it probably depends heavily on gstreamer and the underlying libraries.  I think you could adapt the MJPEG sample code to achieve that.

tomcatxx, 2020-09-15:
Ty very much. For monitoring the detections it fits my needs. I will test it next days.
For full 4K stream opencv needs anyways to much CPU so I just detect with FullHD and record the 4k stream with Zoneminder if detections happens.

closed 2020-09-22 by jkjung-avt