jkjung-avt/tensorrt_demos #213: using flask API yolov4 tensorRT: [TensorRT] ERROR: ../rtSafe/safeContext.cpp (105) - Cudnn Error in initializeCommonContext:

Papageno2, 2020-09-02:
Does anyone has experience about this trt-engine loading with flask api,

 the following code works well if I am not using flask api
```python
# .....
# ... codes from `trt_yolo.py`
# .....

if __name__ == "__main__":
    import pdb
    import os
    os.environ['CUDA_VISIBLE_DEVICES'] = "5"

    args = parse_args()

    cls_dict = get_cls_dict(args.category_num)
    # load model
    yolo_dim = args.model.split('-')[-1]
    if 'x' in yolo_dim:
        dim_split = yolo_dim.split('x')
        if len(dim_split) != 2:
            raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)
        w, h = int(dim_split[0]), int(dim_split[1])
    else:
        h = w = int(yolo_dim)
    if h % 32 != 0 or w % 32 != 0:
        raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)

    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
    conf_th=0.3
    vis = BBoxVisualization(cls_dict)

    for i in range(10):
        t0 = time.time()
        img = cv2.imread(args.input)
        boxes, confs, clss = trt_yolo.detect(img, conf_th)
        print(boxes, confs, clss)
        t1 = time.time()
        print(t1-t0)

    img = vis.draw_bboxes(img, boxes, confs, clss)
    cv2.imwrite('thisis_test_.jpg', img)
```

BUt the error occurs when I use flask api,
```boxes, confs, clss = trt_yolo.detect(img, conf_th)```
 
```python
# .....
# ... codes from `trt_yolo.py`
# .....

@app.route("/object_detection", methods=['POST'])
def detect():
    start = time.time()
    base64_from_post = request.json.get("IMG_BASE64")
    # default： single thres for all classes, float
    # you can pass [thres1, thres2] for two classes
    score_thres = request.json.get("SCORE_THRES", 0.3)
    
    if base64_from_post is not None:
        img_raw = base64toImageArray(base64_from_post) 
        # call trt_yolo for a single detect as in the no-flask-api case
        boxes, confs, clss = trt_yolo.detect(img_raw, score_thres)
        print(clss)
        results = post_process(boxes, confs, clss, score_thres)
      
        results.update({"Errors":0})
        ret = {"predictions": results}
        print("Time spent handling the request: %f" % (time.time() - start))

        return Response(json.dumps(ret, ensure_ascii=False),
                            mimetype='application/json')
    else:
        ret={"predictions":{"Errors": "输入图片参数不能为空!"}}
        return Response(json.dumps(ret, ensure_ascii=False),
                        mimetype='application/json')

if __name__ == "__main__":
    import pdb
    import os
    os.environ['CUDA_VISIBLE_DEVICES'] = "5"

    args = parse_args()

    cls_dict = get_cls_dict(args.category_num)
    # load model
    yolo_dim = args.model.split('-')[-1]
    if 'x' in yolo_dim:
        dim_split = yolo_dim.split('x')
        if len(dim_split) != 2:
            raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)
        w, h = int(dim_split[0]), int(dim_split[1])
    else:
        h = w = int(yolo_dim)
    if h % 32 != 0 or w % 32 != 0:
        raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)

    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)

  # print('Starting the API')
    app.run(debug=False, port=12345, host='0.0.0.0')
```

the output will be,

```shell
root@292264c5f846:/yolov4-darknet-trt/deploy-standalone# python3 deploy_flaskAPI.py -m yolov4-416
[TensorRT] ERROR: ../rtSafe/safeContext.cpp (105) - Cudnn Error in initializeCommonContext: 4 (Could not initialize cudnn, please check cudnn installation.)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
  File "/yolov4-darknet-trt/deploy-standalone/utils/yolo_with_plugins.py", line 253, in __init__
    self.context = self.engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "deploy_flaskAPI.py", line 128, in <module>
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
  File "/yolov4-darknet-trt/deploy-standalone/utils/yolo_with_plugins.py", line 259, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f04145ccb38>>
Traceback (most recent call last):
  File "/yolov4-darknet-trt/deploy-standalone/utils/yolo_with_plugins.py", line 266, in __del__
    del self.outputs
AttributeError: outputs
```

jkjung-avt, 2020-09-02:
Same issue as this one: https://github.com/jkjung-avt/tensorrt_demos/issues/68

I think It's related to managing CUDA context in the thread created by Flask.

jkjung-avt, 2020-09-14:
Could you help to check if this fixes the problem?

Possible solution:  Initialize the TrtYOLO object by assigning the default cuda context to it.  That is, replace this line:

```
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
```

with this:

```
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, cuda_ctx=pycuda.autoinit.context)
```

Papageno2, 2020-09-14:
> Could you help to check if this fixes the problem?
>
> Possible solution: Initialize the TrtYOLO object by assigning the default cuda context to it. That is, replace this line:
>
> ```
>     trt_yolo = [...]

no, it doesn't work with the same errors.

Papageno2, 2020-09-14:
> > Could you help to check if this fixes the problem?
> > Possible solution: Initialize the TrtYOLO object by assigning the default cuda context to it. That is, replace this line:
> > ```
> >     trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
> > ```
> > 
> > 
> > with this:
> > ```
> >     trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, cuda_ctx=pycuda.autoinit.context)
> > ```
> 
> well, it works, thx !!!
> 
> I will give the results of code here soon.

------------ API codes -----------
```python
#  -*- coding: utf-8 -*-

import os
import time
import argparse
import numpy as np
import json
# import tensorflow as tf
from flask import Flask, request, Response
import base64
import cv2
from PIL import Image

import pycuda.autoinit  # This is needed for initializing CUDA driver
from utils.yolo_classes import get_cls_dict
from utils.visualization import BBoxVisualization
from utils.yolo_with_plugins import TrtYOLO

from utils.yolo_class import dictMap_EN 

app = Flask(__name__)

def base64toImageArray(img_base64):
    img_data = base64.b64decode(img_base64)
    image_np = np.frombuffer(img_data, np.uint8)
    # BGR-mode for detecting
    image_np = cv2.imdecode(image_np, cv2.IMREAD_COLOR)
    return image_np

def post_process(boxes, confs, clss, score_thres):
    """
    Note:
    """
    items =[]   
    cnt = 0
    for box, score, cls_ in zip(boxes, confs, clss):
        if (score>score_thres):
            item = {"box": box.tolist(),
                    "score": str(score),
                    "class": dictMap_EN[int(cls_)]
                   }
            items.append(item)
            cnt+=1

    result = {"Nums":cnt,
              "Items":items}    
    return result
    
@app.route("/detect", methods=['POST'])
def detect():
    start = time.time()
    base64_from_post = request.json.get("IMG_BASE64")
    score_thres = request.json.get("SCORE_THRES", 0.3)
    
    if base64_from_post is not None:
        img_raw = base64toImageArray(base64_from_post) 
        boxes, confs, clss = trt_yolo.detect(img_raw, score_thres)
        print(clss)
        results = post_process(boxes, confs, clss, score_thres)
      
        results.update({"Errors":0})
        ret = {"predictions": results}
        print("Time spent handling the request: %f" % (time.time() - start))
        
        return Response(json.dumps(ret, ensure_ascii=False),
                            mimetype='application/json')
    else:
        ret={"predictions":{"Errors": "Inputs cannot be empty!"}}
        return Response(json.dumps(ret, ensure_ascii=False),
                        mimetype='application/json')

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time object detection with TensorRT optimized '
            'YOLO model on Jetson')
    parser = argparse.ArgumentParser(description=desc)
    parser.add_argument(
        '-c', '--category_num', type=int, default=80,
        help='number of object categories [80]')
    parser.add_argument(
        '-m', '--model', type=str, required=True,
        help=('yolov4-'
              '[{dimension}], where dimension could be a single '
              'number (e.g. 416,512,608) or WxH (e.g. 416x256)'))

    parser.add_argument(
        '-i', '--input', type=str,help=('input image path'))
    args = parser.parse_args()
    return args

if __name__ == "__main__":
# USAGE:
# CUDA_VISIBLE_DEVICES=0 python3 xxx.py -m yolov4-512

    import pdb
    import os
    args = parse_args()
    
    cls_dict = get_cls_dict(args.category_num)
    # load model
    yolo_dim = args.model.split('-')[-1]
    if 'x' in yolo_dim:
        dim_split = yolo_dim.split('x')
        if len(dim_split) != 2:
            raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)
        w, h = int(dim_split[0]), int(dim_split[1])
    else:
        h = w = int(yolo_dim)
    if h % 32 != 0 or w % 32 != 0:
        raise SystemExit('ERROR: bad yolo_dim (%s)!' % yolo_dim)

    # pdb.set_trace()

    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, cuda_ctx=pycuda.autoinit.context)
    # trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)    
    conf_th=0.3
    vis = BBoxVisualization(cls_dict)

    print('Starting the API')
    app.run(debug=True, port=11222, host='0.0.0.0')
```

outputs:

```shell
 * Debugger is active!
 * Debugger PIN: 292-650-237
[16. 16. 16. 16.  0.  0.  0. 26.]
Time spent handling the request: 0.048166
172.17.0.1 - - [14/Sep/2020 09:32:25] "POST /detect HTTP/1.1" 200 -
[16. 16. 16. 16.  0.  0.  0. 26.]
Time spent handling the request: 0.039152
172.17.0.1 - - [14/Sep/2020 09:32:26] "POST /detect HTTP/1.1" 200 -
[16. 16. 16. 16.  0.  0.  0. 26.]
Time spent handling the request: 0.042357
172.17.0.1 - - [14/Sep/2020 09:32:27] "POST /detect HTTP/1.1" 200 -
```

it works well !

closed 2020-09-14 by Papageno2

Papageno2, 2020-09-18:
I put the `Dockerfile` for building `image` and deploy_flaskAPI code [here](https://github.com/Papageno2/Dockerfile_collection/tree/master/yolov4-trt/deploy).  
All you need is to get the `yolov4-416.trt` in folder `model/` before building `image`.