jkjung-avt/tensorrt_demos #214: problem about pycuda

Shame-fight, 2020-09-03:
I ran install_pycuda.sh under the ssd folder,but a problem occurred. I do not know how to solve this problem,can you help me ?:
`pycuda-2019.1.2/test/test_gpuarray.py
************************************************************
*** I have detected that you have already run configure.
*** I'm taking the configured values as defaults for this
*** configure run. If you don't want this, delete the file
*** ./siteconf.py.
************************************************************
Usage: configure.py [options]

configure.py: error: no such option: --no-use-shipped-boost`

Shame-fight, 2020-09-03:
@jkjung-avt

jkjung-avt, 2020-09-03:
Just remove the file "${HOME}/src/pycuda-2019.1.2/siteconf.py" and run `python3 ./configure.py ...` again.  Reference:

https://github.com/jkjung-avt/tensorrt_demos/blob/db43bad9050c07723704d53a97b9a4229f09aae4/ssd/install_pycuda.sh#L36

Then you should be able to do `make` and use `python3 setup.py` to build and install pycuda.

Alternatively, just remove the whole directory tree of "${HOME}/src/pycuda-2019.1.2" and run "install_pycuda.sh" from scratch.

Shame-fight, 2020-09-04:
> Just remove the file "${HOME}/src/pycuda-2019.1.2/siteconf.py" and run `python3 ./configure.py ...` again. Reference:
Thanks for your replay,I can run trt_yolo.py now.It can be 4-5 FPS of yolov4-416.But there is a problem:when I run the
 `python3 trt_yolo.py --usb 0 --width 640 --height 480 -m yolov3-416 --category_num 4`,a warning appears: 
`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1`
The video window is dectected normally,but cannot be closed normally.I have modified 'yolo_class.py'like this:
`COCO_CLASSES_LIST = [
    'golf',
    'flag',
    'net',
    'rangeb',
]

# For translating YOLO class ids (0~79) to SSD class ids (0~90)
yolo_cls_to_ssd = [
    1, 2, 3, 4,
]

def get_cls_dict(category_num):
    """Get the class ID to name translation dictionary."""
    if category_num == 4:
        return {i: n for i, n in enumerate(COCO_CLASSES_LIST)}
    else:
        return {i: 'CLS%d' % i for i in range(category_num)}`
I am not sure what caused the problem.by the way ,I am using opencv that comes with jetpack4.4.

jkjung-avt, 2020-09-04:
You could hit ESC to quit the program.

Shame-fight, 2020-09-04:
> You could hit ESC to quit the program.

I cannot.hitting ESC have no response.Ctrl+C sometimes can end it , but it prompts Aborted   `(core dumped).` Sometimes the screen will freeze

jkjung-avt, 2020-09-04:
I don't see such a problem when testing the code on my Jetson Nano and Xavier NX.

What if you try other video sources, such as an image or video file?

Shame-fight, 2020-09-04:
> I don't see such a problem when testing the code on my Jetson Nano and Xavier NX.
> 
> What if you try other video sources, such as an image or video file?

I'm sorry, it's my operation problem. The mouse is placed on the terminal instead of the video causing this problem. So the only problem is the GStreamer warning.
As far as I know, GStreamer has lower latency for reading video streams than opencv. I prefer the former, but I haven't found a solution yet. If you have any suggestions, please let me know.
By the way: I used pip3 to install protobuf and pycuda, which can run yolo_trt on jetson nano well, which saves a lot of time compared to scripts. Thank you again for your work.

closed 2020-09-07 by jkjung-avt