jkjung-avt/tensorrt_demos #216: Running yolov4 demo on onboard camera- tx2

skravoof, 2020-09-05:
Hi,
I was able to successfully run the following command and saw the yolov4 and its working fine. 
 python3 trt_yolo.py --image ${HOME}/Pictures/dog.jpg \  -m yolov4-416
I would like to know, how can test the demo through onboard camera i tried the following but it didnt work.
 python3 trt_yolo.py --onboard 0 -m yolov4-416
Can you please advise
Regards,

jkjung-avt, 2020-09-05:
Are you testing the code on a Jetson TX2?  What error message do you see?

skravoof, 2020-09-05:
I am using Jetson TX2 with jetpack 4.4. I got an error as follows
    raise RuntimeError('no camera type specified!')
RuntimeError: no camera type specified!
I have changed the code from python3 trt_yolo.py --onboard 1 -m yolov4-416    it worked fine.

closed 2020-09-05 by skravoof