jkjung-avt/tensorrt_demos #217: Accessing the Real time video feed from Jetson Nano at a distance of 200 meters!

Farjad3253, 2020-09-05:
Hello! I am in a issue that is I am currently successfully running a custom YOLOv4 model on my Jetson Nano in real time using a logitech C920 pro USB webcam but now I want to send this video in real time to a control room that is 200 meters away from the location where Jetson Nano is placed.
So is there any way to solve this problem?

Shame-fight, 2020-09-07:
As far as I know,If you can get rtsp output, then you can realize data stream transmission through gstreamer.

Farjad3253, 2020-09-07:
@Shame-fight . Thanks for your help but kindly if you can elaborate the point?

jkjung-avt, 2020-09-14:
I created a MJPEG sample.  You could run it and view detection results remotely using a web browser.  Please check https://github.com/jkjung-avt/tensorrt_demos/issues/226#issuecomment-691689456.

Farjad3253, 2020-09-15:
> created a MJPEG sample. You could run it and view detection results remotely using a web browser

Thanks @jkjung-avt . I will give it a try but can you tell that in #226 you have said to check the output in a web browser like : (http://<ip addr>:8080) but what is this ip addr? 

Secondly tell me that the output in the web browser will be a video? And is this web address is local or global? I mean any body sitting any where can access the output or not?

jkjung-avt, 2020-09-15:
> I will definitely give it a try but can you tell that in #226 you have said to check the output in a web browser like : (http://\<ip addr\>:8080) but what is this ip addr?

Say, you run trt_yolo_mjpeg.py on a Jetson Nano with IP address 192.168.0.100, and you'd like to view detection output remotely  using a web browser on a PC.  Then the IP address is 192.168.0.100.

> Secondly tell me that the output in the web browser will be a video?

It's a video compressed in [Motion JPEG](https://en.wikipedia.org/wiki/Motion_JPEG) format.

> And is this web address is local or global?

Yes.

Farjad3253, 2020-09-15:
> Say, you run trt_yolo_mjpeg.py on a Jetson Nano with IP address 192.168.0.100, and you'd like to view detection output remotely using a web browser on a PC. Then the IP address is 192.168.0.100.

Thanks @jkjung-avt  for quick reply and what I understood is that when I run trt_yolo_mjpeg.py then the global server at which I will see the output will have the default port no of 8080 and the ip address will be of my Jetson Nano network i.e. ethernet or WIFI! Is it so, Am I saying right?

jkjung-avt, 2020-09-15:
Yes

Farjad3253, 2020-09-15:
Ok @jkjung-avt . Thanks for the clarification. I will try it out tomorrow and will let you know in case of any error.

closed 2020-09-22 by jkjung-avt

dvitrano, 2020-12-28:
Very interesting "variant" and easy way to share inference results over streaming. @jkjung-avt is there a way to control the MJPEG bitrate? the default is around 15Mbit/s for me. I found HD stream are still ok even with 50% of that bandwidth. Thanks !

jkjung-avt, 2020-12-29:
https://github.com/jkjung-avt/tensorrt_demos/issues/311#issuecomment-751941139