jkjung-avt/tensorrt_demos #218: Met a problem with Downloading cfg file and weights

ZhenshengWu, 2020-09-06:
When i run  ./download_yolo.sh  
I got none response ,there was no change in my file and shell command.And i tried run "chmod u+x download_yolo.sh but it also did not work.
What should i do to solve this problem.Thank you very much.

jkjung-avt, 2020-09-07:
I think this is likely due to some temporary internet connection problems.  You could re-try it later on.

Otherwise, what error message did you see?

ZhenshengWu, 2020-09-07:
Thank you for you promote reply.However, i met another question ,as you know,that in the cfg file of yolov4-tiny ,there were three models --- [route_lhalf] layers=-1. And when i transformed yolo to onnx,it will got an error.It was from the function of "_get_previous_node_specs(self, target_index=0)  assert previous_node.created_onnx_node"  and i know that was because the supported layers  were  ['net', 'convolutional', 'maxpool', 'shortcut', 'route', 'upsample', 'yolo']. Then  i del the three models in cfg file ,but  another error appeared. So,I want to know how i can deal with " [route_lhalf] " Except your reply ,thanks a lot!

jkjung-avt, 2020-09-07:
Take a look at my [dwonload_yolo.sh](https://github.com/jkjung-avt/tensorrt_demos/blob/21ab0e379be796bd00736edac62d466dfba7530e/yolo/download_yolo.sh) script.  I'm using [AlexeyAB/darknet/cfg/yolov4-tiny.cfg](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-tiny.cfg) and it works fine.  There is no "[route_lhalf]" in that cfg file.

Are you not using the official yolov4-tiny.cfg?  If not, which one are you using?

ZhenshengWu, 2020-09-07:
It is really kind of you. After seeing your reply i have found that i get a wrong yolov4-tiny.cfg file which is not come from official.Now ,i have solved this problem.Thanks a lot!

closed 2020-09-07 by jkjung-avt