jkjung-avt/tensorrt_demos #219: how to load more trt model

JF-Lee, 2020-09-07:
Thanks for your working! I have successfully run my yolov4-tiny trt model on NX, but when I want to load 2 models, an error occurred as
follows:

[TensorRT] ERROR: ../rtSafe/cuda/caskConvolutionRunner.cpp (491) - Cuda Error in execute: 400 (invalid resource handle)
[TensorRT] ERROR: FAILED_EXECUTION: std::exception

I hope to get your answer, thank you!!!

jkjung-avt, 2020-09-08:
If you are writing single-threaded code, just instantiate 2 TrtYOLO objects.  It should work.

In case you are creating multiple threads, you'll have to create/manage CUDA context for each thread.  You could refer to [trt_ssd_async.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd_async.py).

JF-Lee, 2020-09-08:
Thanks for your reply, I have tried to instantiate 2 TrtYOLO objects with single-threaded, however It didn't work.
Code and error message are shown in the file.

jkjung-avt, 2020-09-08:
> Code and error message are shown in the file.

Where is the file?

JF-Lee, 2020-09-08:
The file was sent as an attachment, and the content of the file is as follows.

#########################
"""trt_yolo.py
This script demonstrates how to do real-time object detection with
TensorRT optimized YOLO engine.
"""

import os
import time
import argparse
import numpy as np
import cv2
import pycuda.autoinit  # This is needed for initializing CUDA driver
from utils.yolo_classes import get_cls_dict
from utils.yolo import TrtYOLO
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

coco_cls_num = 80
road_cls_num = 1
coco_model = 'yolov4-tiny-416-coco'
road_model = 'yolov4-tiny-416-road'
cap = cv2.VideoCapture('/home/nx/LJF/video/sign.mp4')

if __name__ == '__main__':
    coco_trt = TrtYOLO(coco_model, (416, 416), coco_cls_num)
    road_trt = TrtYOLO(road_model, (416, 416), road_cls_num)
    tic = time.time()
    vid_writer = None
    f_count = 0
    while True:
        ret, frame = cap.read()
        if frame is None:
            continue
        coco_boxes, coco_confs, coco_clss = coco_trt.detect(frame)
        road_boxes, road_confs, road_clss = road_trt.detect(frame)
        print(f_count)
        f_count += 1
        cv2.waitKey(1)
    cap.release()

#########################

jkjung-avt, 2020-09-20:
The following sample code works.  It demonstrates running 2 TrtYOLO detectors in one process.

```
import cv2
import pycuda.autoinit

from utils.yolo_with_plugins import TrtYOLO

img = cv2.imread('dog.jpg')
trt_yolo1= TrtYOLO('yolov4-416', (416, 416), 80)
trt_yolo2= TrtYOLO('yolov4-tiny-288', (288, 288), 80)
print(trt_yolo1.detect(img))
print(trt_yolo2.detect(img))
```

JF-Lee, 2020-09-20:
Thanks for your great work, I have successfully deployed mult trt models with single-thread and mult-thread.Thanks again!

closed 2020-09-20 by jkjung-avt