jkjung-avt/tensorrt_demos #220: How to connect an IP camera with Jetson Nano in place of USB camera?

Farjad3253, 2020-09-07:
Hello @jkjung-avt . Kindly if you can help me in solving the issue of connecting an IP camera with Jetson Nano for doing real time custom YOLOv4 object detection using TensorRT?

jkjung-avt, 2020-09-08:
Please refer to README.md (Step 5 of Demo 1).

> `--rtsp rtsp://admin:123456@192.168.1.1/live.sdp`: RTSP source, e.g. an IP cam. An optional --rtsp_latency argument could be used to adjust the latency setting in this case.

Farjad3253, 2020-09-08:
Thanks @jkjung-avt . I will run my trt_yolo.py with an IP camera and will let you know if found any errors.

Farjad3253, 2020-09-08:
I have followed the step 5 of demo 1 but getting this error:

farjad@farjad-desktop:~/project/tensorrt_demos$ python3 trt_yolo.py --rtsp --uri rtsp://admin:aaasss111@192.168.23.108:554 --model yolov4-custom-416 --category_num 1 --width 1280 --height 720
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module rtspsrc0 reported: Could not open resource for reading.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
ERROR: failed to open camera!

Now what can I do?

jkjung-avt, 2020-09-08:
Are you using a H.264 IP cam?  Please make sure your RTSP stream could be displayed with gst-launch first.  If not, check first that you have all necessary gstreamer elements installed properly.  Otherwise, check NVIDIA Developer Forum for solutions.  And you might need to modify [open_cam_rtsp()](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L53) accordingly.

For example,

```
$ gst-launch-1.0 -e rtspsrc location="rtsp://admin:aaasss111@192.168.23.108:554" ! rtph264depay ! h264parse ! omxh264dec ! nveglglessink
```

Reference: [Video Decode Examples Using gst-launch-1.0](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Faccelerated_gstreamer.html%23wwpID0E0N10HA)

Farjad3253, 2020-09-08:
I have ran gst-lauch also but again getting the error!

jkjung-avt, 2020-09-08:
Do you mean the gst-launch command also failed?

Farjad3253, 2020-09-08:
@jkjung-avt . Yes Sir. The gst-launch command also failed! The errors which I am getting are:

farjad@farjad-desktop:~$ gst-launch-1.0 rtspsrc location=rtsp://admin:aaasss111@192.168.23.108:554/media/video1 ! rtph264depay  ! avdec_h264 ! videoconvert ! gtksink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:aaasss111@192.168.23.108:554/media/video1
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not open resource for reading.
Additional debug info:
gstrtspsrc.c(5829): gst_rtspsrc_setup_auth (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
No supported authentication protocol was found
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

jkjung-avt, 2020-09-08:
You need to make sure you have a working RTSP stream and decoding pipeline first.  I can't help too much on that.

Farjad3253, 2020-09-08:
> You need to make sure you have a working RTSP stream and decoding pipeline first. I can't help too much on that.

But how to make sure that because I have tried all the necessary commands but not able to get the working RTSP stream?

Farjad3253, 2020-09-08:
> @jkjung-avt . Yes Sir. The gst-launch command also failed! The errors which I am getting are:
>
> farjad@farjad-desktop:~$ gst-launch-1.0 rtspsrc [...]

Can you please point out that what is the error here?

Farjad3253, 2020-09-08:
> farjad@farjad-desktop:~/project/tensorrt_demos$ python3 trt_yolo.py --rtsp --uri rtsp://admin:aaasss111@192.168.23.108:554 --model yolov4-custom-416 --category_num 1 --width 1280 --height 720
> [ [...]
Also what is the error here?

jkjung-avt, 2020-09-08:
Have you made sure you could play the RTSP stream with, say, VLC or QuickTime player?

Reference: http://www.messoa.com/faq_detail.php?i=5

Farjad3253, 2020-09-08:
@jkjung-avt . 

> Have you made sure you could play the RTSP stream with, say, VLC or QuickTime player?
> 
> Reference: http://www.messoa.com/faq_detail.php?i=5

I will try this out and will let you know. Thanks a lot!

closed 2020-09-11 by jkjung-avt

Myron1996, 2021-06-08:
Hi @jkjung-avt . It might be very silly question but can yu please explain 

--rtsp rtsp://admin:123456@192.168.1.1/live.sdp

what exactly is `admin:123456`

do I have to put my admin password in that location ?

Thank you

jkjung-avt, 2021-06-08:
> --rtsp rtsp://admin:123456@192.168.1.1/live.sdp

Most of RTSP sources (e.g. IP CAM) would require an account/password to access the video stream.  You should replace "admin" with the account and "123456" with the password of your own RTSP stream.

Myron1996, 2021-06-08:
Thanks @jkjung-avt .