jkjung-avt/tensorrt_demos #221: How to measure the distance of a target

Shame-fight, 2020-09-08:
Hi, it's still me, I hope I didn't bother you. Do you have any idea of measuring the distance between the target and the camera? Just like AlexeyAB uses the zed binocular camera.
 `ZED_CAMERA=1 to build a library with ZED-3D-camera support (should be ZED SDK installed), then run LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./uselib data/coco.names cfg/yolov4.cfg yolov4.weights zed_camera.` 
This is very important for driverless driving, and I look forward to your reply.

Shame-fight, 2020-09-08:
In other words, how to measure the distance of the detected target with a zed camera and output the coordinates (for example x, y, w, h, z)

jkjung-avt, 2020-09-08:
I think you need to implement some post-processing to combine yolo detections with the 3D output of ZED camera.  Refer to [get_3d_coordinates()](https://github.com/AlexeyAB/darknet/blob/master/src/yolo_console_dll.cpp#L46) in AlexeyAB's darknet code.

Shame-fight, 2020-09-08:
> I think you need to implement some post-processing to combine yolo detections with the 3D output of ZED camera. Refer to [get_3d_coordinates()](https://github.com/AlexeyAB/darknet/blob/master/src/yolo_console_dll.cpp#L46) in AlexeyAB's darknet code.

Yes, but I feel a little bit at a loss, maybe my programming ability is relatively weak, so I come to ask you for help. Please forgive my'not working hard', I'm currently in the preliminary stage. I have a more shameless request, can you Take the time to complete this function. I admire the work you do and I am learning your code.

jkjung-avt, 2020-09-08:
I don't have a ZED camera.  I don't have a plan to work on that either.

Shame-fight, 2020-09-08:
> I don't have a ZED camera. I don't have a plan to work on that either.

ok,thanks

closed 2020-09-08 by Shame-fight