jkjung-avt/tensorrt_demos #222: Getting issue while doing make in plugins

goutamgupta, 2020-09-08:
Hi , 
I am getting the follwoing erro while doing make in plugins folder. Can please help to resolve this erro?

/usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_62,code=compute_62 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
g++: error: yolo_layer.o: No such file or directory
Makefile:50: recipe for target 'libyolo_layer.so' failed
make: *** [libyolo_layer.so] Error 1

Thanks
GOutam

jkjung-avt, 2020-09-08:
Are you using TensorRT-7.1.3.4?  Do make sure the path "/usr/local/TensorRT-7.1.3.4/include" (in the Makefile) is correct for your environment.

Otherwise, please attach the full log of `make` for checking.  I don't see why nvcc fails to build "yolo_layer.o" above...

goutamgupta, 2020-09-08:
I am using Jetpack 4.4 which installs tensorrt 7.1.3.0. I am not able to get the path of the tensorrt installation as it is mentioned in the makefile. I tried using " /usr/src/tensorrt" as the path but the 'lib' and 'include' directories are not present there. How do i find the path of the tensorrt installation if i have installed tensorrt through deepstream sdk on Jetson TX2. 

Thanks
Goutam

jkjung-avt, 2020-09-08:
I see.  For TX2, you don't need to modify the Makefile.  TensorRT include/library should be in the default paths (/usr/includ/aarch64-linux-gnu & /usr/lib/aarch64-linux-gnu) already.

Please execute the nvcc command manually to see if "yolo_layer.o" is generated.

```
/usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_62,code=compute_62 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
```

If not, please add "--verbose" and run the command again.  Attach the log so I could check what might be the problem.

goutamgupta, 2020-09-08:
I tried this command manually. There is no message even after adding --verbose. yolo_layer.o is not generated.

elma@elma-desktop:/media/elma/a3544183-e86d-4ae6-b0fc-a77e32544b55/tensorrt_demos/plugins$ /usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.0/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_62,code=compute_62 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu --verbose
elma@elma-desktop:/media/elma/a3544183-e86d-4ae6-b0fc-a77e32544b55/tensorrt_demos/plugins$ ls
Makefile  README.md  yolo_layer.cu  yolo_layer.h

jkjung-avt, 2020-09-08:
I have no idea why this happens...

Could you copy the whole directory to another location (not on /media/XXX) and do `make` to see if it makes a difference?

goutamgupta, 2020-09-08:
The result is same.
elma@elma-desktop:~/tensorrt_demos/plugins$ /usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.0/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_62,code=compute_62 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu --verbose
elma@elma-desktop:~/tensorrt_demos/plugins$ ls
Makefile  README.md  yolo_layer.cu  yolo_layer.h
elma@elma-desktop:~/tensorrt_demos/plugins$

goutamgupta, 2020-09-08:
elma@elma-desktop:~/tensorrt_demos/plugins$ sudo make
/usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_62,code=compute_62 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
g++: error: yolo_layer.o: No such file or directory
Makefile:50: recipe for target 'libyolo_layer.so' failed
make: *** [libyolo_layer.so] Error 1

jkjung-avt, 2020-09-08:
What are the outputs of `which nvcc` and `nvcc -V`?

And are you able to build official cuda samples with your "nvcc"?  Install cuda samples source code by following the instructions in the link below.  Then try `make` the `1_Utilities/deviceQuery` sample program.  Check if "deviceQuery" could be built and run correctly.

https://docs.nvidia.com/cuda/cuda-samples/index.html#getting-started-with-cuda-samples

goutamgupta, 2020-09-10:
Hi, 
I am not able to make deviceQuery sucessfully. Do i need to uninstall and re-install cuda?

elma@elma-desktop:/usr/local/cuda-10.2/bin/NVIDIA_CUDA-10.2_Samples/1_Utilities/deviceQuery$ which nvcc 
/usr/local/cuda-10.2/bin/nvcc

elma@elma-desktop:/usr/local/cuda-10.2/bin/NVIDIA_CUDA-10.2_Samples/1_Utilities/deviceQuery$ make && ./deviceQuery
/usr/local/cuda-10.2/bin/nvcc -ccbin g++ -I../../common/inc  -m64    -gencode arch=compute_30,code=sm_30 -gencode arch=compute_32,code=sm_32 -gencode arch=compute_53,code=sm_53 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_72,code=sm_72 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o deviceQuery.o -c deviceQuery.cpp
/usr/local/cuda-10.2/bin/nvcc -ccbin g++   -m64      -gencode arch=compute_30,code=sm_30 -gencode arch=compute_32,code=sm_32 -gencode arch=compute_53,code=sm_53 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_62,code=sm_62 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_72,code=sm_72 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -o deviceQuery deviceQuery.o 
mkdir -p ../../bin/aarch64/linux/release
cp deviceQuery ../../bin/aarch64/linux/release
cp: cannot stat 'deviceQuery': No such file or directory
Makefile:303: recipe for target 'deviceQuery' failed
make: *** [deviceQuery] Error 1

jkjung-avt, 2020-09-10:
"nvcc" compiler appears not working on your TX2.  You need to fix that first.  You might try to reinstall cuda toolkit, or just re-flash the TX2 with JetPack-4.4 again.

jkjung-avt, 2020-09-20:
@goutamgupta Did you manage to fix the problem?  Could we close this issue now?

goutamgupta, 2020-09-20:
Hi, 
Yeah, i reflashed the Jetson and things seem to work. Looks like my CUDA installation got corrupted. Sorry, forgot to update on the forum. Thanks for your support. 

Thanks
GOutam

closed 2020-09-20 by jkjung-avt

rodolfobetanzos, 2020-11-23:
Hi jkjung-avt !!1 I am following  your repo 

[https://github.com/jkjung-avt/tensorrt_demos]
(https://github.com/jkjung-avt/tensorrt_demos)
but on the setp 3.- for Demo #5: YOLOv4 where "Go to the "plugins/" subdirectory and build the "yolo_layer" plugin. When done, a "libyolo_layer.so" would be generated." **I get the next error** ..

betanzos@betanzos:~/project/tensorrt_demos/plugins$ make
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/lo

cal/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_53,code=sm_53 -gencode arch=compute_53,code=compute_53 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(38): error: not a class or struct name

yolo_layer.h(44): error: member function declared with "override" does not override a base class member

yolo_layer.h(46): error: member function declared with "override" does not override a base class member

yolo_layer.h(51): error: member function declared with "override" does not override a base class member

yolo_layer.h(53): error: member function declared with "override" does not override a base class member

yolo_layer.h(55): error: member function declared with "override" does not override a base class member

yolo_layer.h(57): error: member function declared with "override" does not override a base class member

yolo_layer.h(59): error: member function declared with "override" does not override a base class member

yolo_layer.h(61): error: member function declared with "override" does not override a base class member

yolo_layer.h(63): error: member function declared with "override" does not override a base class member

yolo_layer.h(65): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(65): error: member function declared with "override" does not override a base class member

yolo_layer.h(69): error: member function declared with "override" does not override a base class member

yolo_layer.h(71): error: member function declared with "override" does not override a base class member

yolo_layer.h(73): error: member function declared with "override" does not override a base class member

yolo_layer.h(75): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(75): error: member function declared with "override" does not override a base class member

yolo_layer.h(77): error: member function declared with "override" does not override a base class member

yolo_layer.h(79): error: member function declared with "override" does not override a base class member

yolo_layer.h(81): error: member function declared with "override" does not override a base class member

yolo_layer.h(83): error: member function declared with "override" does not override a base class member

yolo_layer.h(85): error: member function declared with "override" does not override a base class member

yolo_layer.h(87): error: member function declared with "override" does not override a base class member

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: member function declared with "override" does not override a base class member

yolo_layer.h(89): error: expected a ";"

yolo_layer.h(91): error: member function declared with "override" does not override a base class member

yolo_layer.h(66): error: name followed by "::" must be a class or namespace name

yolo_layer.h(120): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(120): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::createPlugin"

yolo_layer.h(120): error: member function declared with "override" does not override a base class member

yolo_layer.h(122): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(122): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::deserializePlugin"

yolo_layer.h(122): error: member function declared with "override" does not override a base class member

yolo_layer.cu(151): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(151): error: identifier "PluginTensorDesc" is undefined

yolo_layer.cu(181): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(282): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(347): error: identifier "IPluginV2IOExt" is undefined

/usr/include/aarch64-linux-gnu/NvInfer.h(5935): error: object of abstract class type "nvinfer1::YoloPluginCreator" is not allowed:
            pure virtual function "nvinfer1::IPluginCreator::createPlugin" has no overrider
            pure virtual function "nvinfer1::IPluginCreator::deserializePlugin" has no overrider
          detected during instantiation of class "nvinfer1::PluginRegistrar<T> [with T=nvinfer1::YoloPluginCreator]" 
yolo_layer.cu(356): here

41 errors detected in the compilation of "/tmp/tmpxft_0000135a_00000000-6_yolo_layer.cpp1.ii".
Makefile:48: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

Please your support !!! I will be very grateful

jkjung-avt, 2020-11-24:
https://github.com/jkjung-avt/tensorrt_demos/issues/230#issuecomment-694583342
https://github.com/jkjung-avt/tensorrt_demos/issues/156#issuecomment-691173417