jkjung-avt/tensorrt_demos #224: about yolov3 and yolov4->onnx->trt

bamboosdu, 2020-09-09:
Hi, thanks for your wonderful work, and I have a question that whether the code for turning yolov3(darknet or pytorch) to trt is same with the code for turning yolov4(darknet or pytorch) or not.
If not, what is the difference？Looking forward to your reply！

jkjung-avt, 2020-09-09:
Yes.

The code in this repo can be used to convert darknet yolo models (cfg/weights) into onnx models and TensorRT engines.  Pytorch is not supported.  The code is exactly the same for yolov3 and yolov4 models.

bamboosdu, 2020-09-10:
> Yes.
> 
> The code in this repo can be used to convert darknet yolo models (cfg/weights) into onnx models and TensorRT engines. Pytorch is not supported. The code is exactly the same for yolov3 and yolov4 models.

Thanks a  lot for  your kind reply!

closed 2020-09-11 by jkjung-avt