jkjung-avt/tensorrt_demos #225: How to raise an external alarm once object is detected using YOLOv4!

Farjad3253, 2020-09-09:
Hello everyone! I want to add something that if I add some lines of code in trt_yolo.py , I mean more specifically if I add some lines of code inside the loop_and_detect() function after line number 63 that:

if confs > 0.2 :
    GPIO.output(Pin Number 13 of Jetson Nano)

So, anybody can help me that if I am saying right?

joelosw, 2020-09-09:
I have not looked at it in depth, but I would guess: 
1. confs seems like a list to me. So you need to check if there is at least one value >0.5 (or whatever confidence you want) is included. But be aware, that results with conf of <0.3 are already filtered out in trt_yolo.detect(). 
2. https://github.com/NVIDIA/jetson-gpio this Repo seems to be pretty much exactly what you want. 
You'll have to do the setup once in the beginning and can then set the GPIO activation once an object is detected with significant confidence.

Farjad3253, 2020-09-09:
Thanks @joelosw  for your help! I will try and will let you know for any errors.

closed 2020-09-22 by jkjung-avt