jkjung-avt/tensorrt_demos #226: Jetson nano - Run yolo code as headless (without GUI)

dhgokul, 2020-09-09:
@jkjung-avt  Thanks for your sample code, its very helpful !

I want to run yolo code as headless (without GUI) . Need to write as output videos/image . Please provide option to run yolo code to print console log instead of display in screen

I have jetson nano board without HDMI display connect . i want to run program and write it as output file

Any help appreciated! 
cmd :  python3 trt_yolo.py --image dog.jpg  -m yolov3-tiny-416

[image]

jkjung-avt, 2020-09-13:
I just created a [trt_yolo_mjpeg.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_yolo_mjpeg.py) sample code.  Please `git pull` the latest code from the repo and give it a try.  For example, you could run trt_yolo_mjpeg.py headlessly like this:

```
$ python3 trt_yolo_mjpeg.py --usb 0 --copy_frame -m yolov4-416
```

Then check out the output in a web browser (http://\<ip addr\>:8080).

By default, the program listens to port 8080 and publishes MJPEG streams.  The MJPEG image is refreshed roughly every 0.1 second.  You could use "--mjpeg_port" command-line option to change the MJPEG server port.

I haven't found a good way to terminate the program.  So currently you have to hit Ctrl-C a couple of times to force the program to stop.  This should probably be improved in the future.

dhgokul, 2020-09-14:
@jkjung-avt  Thanks! will check

closed 2020-09-22 by jkjung-avt

Enso-Zheng, 2021-08-11:
very useful! thnaks u!!