jkjung-avt/tensorrt_demos #227: (Error on NX):yolo_layer.o: error adding symbols

CODEDISON, 2020-09-10:
Error occurs when I tried to run make under /tensorrt_demos/plugins.
I have already install the prerequisites,pycuda and onnx.

`
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
/usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
yolo_layer.o: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
Makefile:45: recipe for target 'libyolo_layer.so' failed
make: *** [libyolo_layer.so] Error 1

`

jkjung-avt, 2020-09-10:
> /usr/bin/ld: yolo_layer.o: Relocations in generic ELF (EM: 62)
> yolo_layer.o: error adding symbols: File in wrong format

This means your "yolo_layer.o" is in a wrong format.

Please check whether the problem persists if you do `make clean; make`.

CODEDISON, 2020-09-10:
Thx for reply!The problem is solved.But when I run onnx_to_tensorrt.py,this shows up
`
Loading ONNX file from path yolov4-tiny-288.onnx...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" to enable verbose logging.)
[TensorRT] ERROR: 004_route: out of bounds slice, input dimensions = [1,64,72,72], start = [0,0,0,0], size = [8,32,72,72], stride = [1,1,1,1].
[TensorRT] ERROR: Layer 004_route failed validation
[TensorRT] ERROR: Network validation failed.
Completed creating engine.
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 131, in <module>
    main()
  File "onnx_to_tensorrt.py", line 126, in main
    f.write(engine.serialize())
AttributeError: 'NoneType' object has no attribute 'serialize'

`

CODEDISON, 2020-09-10:
I trained my own data with 3 classes,using yolov4-tiny-288.cfg.

jkjung-avt, 2020-09-10:
Please set `bacth=1` in your cfg file.

That is, modify this line in the cfg: [yolov4-tiny.cfg](https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-tiny.cfg#L6)

CODEDISON, 2020-09-10:
It works!Much Obliged!

closed 2020-09-11 by jkjung-avt

adrianosantospb, 2020-09-11:
> Thx for reply!The problem is solved.But when I run onnx_to_tensorrt.py,this shows up
> `
> Loading ONNX file from path yolov4-tiny-288.onnx...
> Adding yolo_layer plugins...
> Building an engine. This would take a while...
> (Use "--verbose" to enable verbose logging.)
> [TensorRT] ERROR: 004_route: out of bounds slice, input dimensions = [1,64,72,72], start = [0,0,0,0], size = [8,32,72,72], stride = [1,1,1,1].
> [TensorRT] ERROR: Layer 004_route failed validation
> [TensorRT] ERROR: Network validation failed.
> Completed creating engine.
> Traceback (most recent call last):
> File "onnx_to_tensorrt.py", line 131, in
> main()
> File "onnx_to_tensorrt.py", line 126, in main
> f.write(engine.serialize())
> AttributeError: 'NoneType' object has no attribute 'serialize'
> 
> `
How did you solve it? I have the same issue.

jkjung-avt, 2020-09-11:
> How did you solve it?

https://github.com/jkjung-avt/tensorrt_demos/issues/227#issuecomment-690110242

Otherwise, I just made [this commit](https://github.com/jkjung-avt/tensorrt_demos/commit/82dbbacc86cf558b1f70fca938f0e13130b281f2) to force batch_size=1 in the ONNX file.  So if you `git pull` the latest code, it should just work.

adrianosantospb, 2020-09-11:
@jkjung-avt , I did this. I'm using the YoloV3 (Tiny3) - (https://github.com/ultralytics/yolov3/blob/master/cfg/yolov3-tiny3-1cls.cfg) 
This version uses 3 YOLO layers ( 16, 23, and 30). Different anchors as well (9 instead 6), 3 yolo_masks. I added a line on output_tensor_dims as well (32, 16, 8)