jkjung-avt/tensorrt_demos #231: Need output image in Full HD resolution.

goutamgupta, 2020-09-19:
Hi, 
The camera that i am using is capable of capturing image in Full HD resolution. What do we need to do to save the inference/detection output in the Full HD resolution?  I am using Yolov4-608 model and that is generating output of 608x608. Open cv's image resize function makes the output very blurry. I tried opencv super resolution package but that makes the inference speed too slow. Is there any way so that the output image resolution is same as the input image resolution. 

Thanks
Goutam

jkjung-avt, 2020-09-20:
My [camera code](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L46) resizes input images to 640x480 by default.  Please check if adding "--width 1920 --height 1080" command-line options suits your need.

goutamgupta, 2020-09-20:
Adding width --1920 --height 1080 only changes the input image size. I am looking for the output image i.e the image with detect objects(bounding box) to be in 1920x1080size. Image resize option makes the image too blury. Is there any option to make it? Does YoloV4 model accepts image of Full HD resolution?
Thanks
Goutam

jkjung-avt, 2020-09-20:
YOLOv4 could handle any input image dimension as long as both the width and height are multiples of 32.

For example, you could copy the original yolov4.cfg and create a yolov4-1920x1088.cfg (remember to set "width=1920" and "height=1088" in the file as well).  Then build a new TensorRT engine with it.  Note that such a TensorRT engine would require more memory and run much more slowly than the 608x608 one.

goutamgupta, 2020-09-20:
Thanks. Can you please elaborate more on this. I mean , how to create a yolov4-1920x1080.cfg and build new TensorRT. Is there any tutorial that i can refer to ?
Thanks
Goutam

jkjung-avt, 2020-09-20:
Please refer to what I did in [download_yolo.sh](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolo/download_yolo.sh).

You make a copy of the original yolov4.cfg and yolov4.weights.  Rename the files as yolov4-1920x1088.cfg and .weights.  (Note it's 1088, not 1080.)  You'd also modify the "width" and "height" in the cfg file accordingly.  Then you could use the newly created .cfg and .weights to build a yolov4-1920x1088 TensorRT engine.

goutamgupta, 2020-09-21:
I followed following steps: 
1. Added these line to download_yolo.sh
echo "Creating yolov4-1088.cfg and yolov4-1088.weights"
cat yolov4.cfg | sed -e '2s/batch=64/batch=1/' | sed -e '7s/width=608/width=1920/' | sed -e '8s/height=608/height=1088/' > yolov4-1920x1088.cfg
ln -sf yolov4.weights yolov4-1920x1088.weights

2. Ran following command to generate trt engine 
sudo python3 yolo_to_onnx.py -m yolov4-1920x1088 
sudo python3 onnx_to_tensorrt.py -m yolov4-1920x1088 

3. yolov4-1920x1088.trt is generated successfully. 
4. Ran following command 
sudo python3 trt_yolo.py --video=/dev/video2 --width 1920 --height=1080 -m yolov4-1920x1088
but i am getting following error : 

[ WARN:0] global /mnt/opencv-4.4.0/modules/videoio/src/cap_gstreamer.cpp (1761) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /mnt/opencv-4.4.0/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /mnt/opencv-4.4.0/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "trt_yolo.py", line 116, in <module>
    main()
  File "trt_yolo.py", line 109, in main
    loop_and_detect(cam, trt_yolo, conf_th=0.3, vis=vis)
  File "trt_yolo.py", line 63, in loop_and_detect
    boxes, confs, clss = trt_yolo.detect(img, conf_th)
  File "/mnt/tensorrt_demos/utils/yolo_with_plugins.py", line 284, in detect
    stream=self.stream)
  File "/mnt/tensorrt_demos/utils/yolo_with_plugins.py", line 203, in do_inference_v2
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
  File "/mnt/tensorrt_demos/utils/yolo_with_plugins.py", line 203, in <listcomp>
    [cuda.memcpy_htod_async(inp.device, inp.host, stream) for inp in inputs]
pycuda._driver.LogicError: cuMemcpyHtoDAsync failed: invalid argument

5. but this runs successfully : 
sudo python3 trt_yolo.py --video=/dev/video2 --width 1920 --height=1080 -m yolov4-416

Can you please help to resolve this?
Thanks
Goutam

goutamgupta, 2020-09-21:
There is one more issue that i am facing. It seems that --width=1920 and --height=1080 is not being read properly from the command line. I can see  parser = add_camera_args(parser) in the code which should in effect reflect this change. But if i print(cam.img_width and cam.img_height) it shows the default values which is 640x480. Is there any way that can be addressed?

jkjung-avt, 2020-09-21:
> There is one more issue that i am facing. It seems that --width=1920 and --height=1080 is not being read properly from the command line. I can see parser = add_camera_args(parser) in the code which [...]

That's most likely due to the following line of code.  When the Camera object tries to read 1 image from the camera, it gets an image of 640x480, not 1920x1080.

https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L208

goutamgupta, 2020-09-21:
Why is it done like that?  Camera Object should read as per the image --width and --height parameter? What modification can we make so that the camera object reads image of 1920x1080. 
Also, can you please comment on the issue that i am facing while generating yolov4-1920x1088.trt file?

jkjung-avt, 2020-09-21:
Are you using a USB camera?  For example, I explicitly ask the camera to output {width}x{height} (1920x1080) images here.  However, if the camera's read() function still returns 640x480 images, I'd reset img_width/img_height to 640x480.

https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/camera.py#L79

closed 2020-09-25 by jkjung-avt