jkjung-avt/tensorrt_demos #236: How to check FPS in eval_yolo.py

hjlee9182, 2020-09-24:
I have a two question.

1. How to check FPS in eval_yolo.py ?

2. How to calculate mAP under image?
[image]

I think IoU=0.50:0.95 is mAP, is it right?

jkjung-avt, 2020-09-24:
> 1. How to check FPS in eval_yolo.py 

"eval_yolo.py" is only for calculating mAP.  You could use "trt_yolo.py" to test FPS.

> 2. How to calculate mAP under image?

Yes.  I'd look at both "IoU=0.50:0.95" and "IoU=0.50".  You could refer to [Verifying mAP of TensorRT Optimized SSD and YOLOv3 Models](https://jkjung-avt.github.io/trt-detection-map/) for more explanations.

hjlee9182, 2020-09-24:
@jkjung-avt 

Really thnk U.

I have anothe question.

what is FP 16 ?  is it FloatPoint?

jkjung-avt, 2020-09-24:
FP16 is referring to half-precision (16-bit) floating-point.  Please read [TensorRT documentation](https://docs.nvidia.com/deeplearning/tensorrt/developer-guide/index.html) to get a better understanding.

closed 2020-09-25 by jkjung-avt