jkjung-avt/tensorrt_demos #237: Detection results darknet vs. trt engine

mosch91-syn, 2020-09-24:
Hi jkjung,

first of all thank you very much for your awesome work. With this repo I can achieve approx. 2 FPS with custom yolov4-608 weights. That said, I am experiencing quite a view detection differences when running with tensorRT compared native darknet. In darknet, all objects even with low confidence threshold of 0.3 are detected. In TensorRT and the trt_yolo.py script, I am getting a lot of "bulked" False positives at confidence up to 0.6. This leads me to setting the confidence thresold in the trt_yolo script to 0.65, in order to only get the right detections. The downside unfortunatelly is, that True Positives at confidence around 0.3 are filtered out. Is there any parameter in the conversion scripts I have to tune in order to filter out bulked false positives in the trt engine? 

Thank you in advance!

jkjung-avt, 2020-09-24:
One possible cause of such a discrepancy is the use of FP16 optimization.  And it's easy to verify if that's the case.  You could try to build a FP32 TensorRT engine by setting `builder.fp16_mode` to `False` in the code.

https://github.com/jkjung-avt/tensorrt_demos/blob/82dbbacc86cf558b1f70fca938f0e13130b281f2/yolo/onnx_to_tensorrt.py#L74

mosch91-syn, 2020-09-24:
Thanks for the quick answer, will try and give feedback!

mosch91-syn, 2020-09-24:
I tried with FP32 model but unfortunately, false positives still remain and dominate below a threshold of 0.66. Any other suggestions to solve this? Thank you in advance!

jkjung-avt, 2020-09-25:
I have just reviewed my own code (yolo_layer cuda kernel and postprocessing).  I think it's doing the exact same computation as what AlexeyAB's [darknet](https://github.com/AlexeyAB/darknet) does.  So I don't quite have an idea right now...

Maybe you could double check whether you do the same image **preprocessing** for training and inferencing, especially on the "aspect ratio" of the images.

jkjung-avt, 2020-09-25:
Thanks for reporting this issue.

Please try modifying this line of code: https://github.com/jkjung-avt/tensorrt_demos/blob/1b274b874b59c0c7ed05f5efb2d89c0089fa3a15/utils/yolo_with_plugins.py#L56

with the following:

```
    box_confidences = detections[:, 4] * detections[:, 6]
```

Let me know if that makes a difference for you.

mosch91-syn, 2020-09-25:
Thanks jkjung, will definetly try and let you know, if it worked!

mosch91-syn, 2020-09-28:
Hi jkjung, I tried but unfortunatelly, results are still the same! Any other ideas? Thank you!

jkjung-avt, 2020-10-02:
Sorry, I'm out of ideas for now.  If I think of something later on, I will let you know.

Alphonses, 2021-01-20:
Hello,@mosch91-syn .You need add a lot of negative sample data to train so that the model can reduce mis-recognition.

jkjung-avt, 2021-03-24:
I'm closing this issue and will continue to track it here: https://github.com/jkjung-avt/tensorrt_demos/issues/315

closed 2021-03-24 by jkjung-avt