jkjung-avt/tensorrt_demos #238: TensorRT yolov3 inference too slowly

taobiaoli, 2020-09-24:
Based on the optimization you provide.
1. float32 -->float16
2.  modify function  “sigmoid” and “exponential”

My test environment is jetson xavier with tensorrt7.  the model is darknet yolov3 and converted onnx.
The Inference time about cost 45ms, and postpropress time about cost 175ms before optimization.
But there is little difference used optimize.
I think post-processing takes too much time,Do you have a better optimization plan to reduce the post-processing time?
I want video detection up to 30 fps. Do you have any suggestions? thank you.

jkjung-avt, 2020-09-24:
You could use python "cProfile" to find the most time-consuming parts of the code and try to optimize them.  I demonstrated that in my [Optimizing TensorRT MTCNN](https://jkjung-avt.github.io/optimize-mtcnn/) blog post.

You mentioned that "postpropress time about cost 175ms".  Are you referring to my "trt_yolo.py" implementation??  That is very different from what I observed on my Jetson Nano and Xavier NX.  The most time-consuming work in postprocessing is NMS.  On my Jetson Nano, I think that part usually takes less than 10ms per image.  (It might take longer if there are a lot of potential objects that need to be calculated by NMS...)

taobiaoli, 2020-09-25:
Thank you for your reply.

I tested the speed of the Yolov3 model  according to the "trt_yolo.py" script you provided,today. The FPS is about 34.

My previous test used "yolov3_onnx.py" that provided by tensorrt. I found that its post-processing is directly implemented with Python, especially in Yolo layer that is too slowly. The "trt_yolo.py" you provided seems to implement Yolo layer with CUDA in "plugins" file.

jkjung-avt, 2020-09-25:
Right, I implemented most of the yolo layer (postprocessing) code in the yolo_layer plugin (CUDA code).  So it would be much faster than NVIDIA's original "yolov3_onnx" sample.

In fact, the postprocessing code in the official TensorRT "yolov3_onnx" sample is quite inefficient.  I discussed it in my [TensorRT ONNX YOLOv3](https://jkjung-avt.github.io/tensorrt-yolov3/) blog post (refer to the "Optimizing the postprocessing function" section).

closed 2020-09-26 by jkjung-avt