jkjung-avt/tensorrt_demos #239: Could Not Detect any of the COCO dataset Objects

NaeemKhan333, 2020-09-24:
@jkjung-avt Thanks for nice work , I have converted yolov3-tiny.weights(yolov3-tiny-288.weights) to yolov3-tiny.trt(yolo-tiny-288.trt).For conversion inTensorRt , I have follow following steps

https://github.com/jkjung-avt/tensorrt_demos#yolov3

But After completing the conversion process, I could not get any detection and bounding box list having empty. Can you guide what I am doing wrong in conversion. I am not getting any error during the conversion.

jkjung-avt, 2020-09-25:
I cannot give much advice based on this description...

Do you mean the "yolov3-tiny-288" TensorRT engine could not detect anything in this [dog.jpg](https://raw.githubusercontent.com/AlexeyAB/darknet/master/data/dog.jpg) image?

NaeemKhan333, 2020-09-25:
Yeah,It does not detect and draw bounding box on dog.jpg. It also does not draw any other image or video. I do not know why but I follow your instruction steps given on github repo.

NaeemKhan333, 2020-09-25:
Secondly , I am not using the jetson-nano for the testing, I have my own GTX-1080 Gpu. Is could be the reason?

jkjung-avt, 2020-09-25:
I've tested the code on my GTX-2080 Ti.  It should work.

jkjung-avt, 2020-10-05:
Closing this issue since it's a duplicate to https://github.com/jkjung-avt/tensorrt_demos/issues/240.

closed 2020-10-05 by jkjung-avt