jkjung-avt/tensorrt_demos #240: Multiple Bounding Boxes For Single Object

NaeemKhan333, 2020-09-24:
I have used one of old commit of your code . It is giving me resultant bounding box on the video input.But it is giving multiple bounding for the single object. Its giving one larger bbox and one small bbox. I do not know why it is giving such input . I have "checkout the following commit " and use the code for getting the output

https://github.com/jkjung-avt/tensorrt_demos/commit/8dbcf2d979b4f4d62075124fe8c4d4c7aad71081

For Conversion, I have follow the link below
https://github.com/jkjung-avt/tensorrt_demos#yolov3

It is giving me the output,but giving two different bounding boxes for the single objects.It drawing multiple bounding boxes i.e A small bbox and a large bbox. can you guide why this problem happens .Thanks

[image]

jkjung-avt, 2020-09-25:
Judging from the picture you've provided, I think you used wrong "anchor boxes" for the detection.

https://github.com/jkjung-avt/tensorrt_demos/blob/8dbcf2d979b4f4d62075124fe8c4d4c7aad71081/utils/yolov3.py#L430

Anyway, I'd suggest you to use the latest code in this repo.

NaeemKhan333, 2020-09-25:
I have used the latest code but it is not giving any detection on this image. So I have checkout the previous commit of your code to  to get the detection . But it is giving the false detection results.

NaeemKhan333, 2020-09-25:
I am testing this code on the my Gpu GTX-1080, not jetson-nano. Is could be the reason for false detection?

jkjung-avt, 2020-09-25:
Please refer to [README_x86.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md) and https://github.com/jkjung-avt/tensorrt_demos/issues/230.

Make sure "plugins/libyolo_layer.so" has been built successfully before you build the TensorRT engine.

NaeemKhan333, 2020-09-25:
is there is difference in installation between jetson nano and Desktop PC?

jkjung-avt, 2020-09-25:
On a Jetson platform, you just need to flash it with JetPack.  But on an x86_64 PC, you'll have to install CUDA Toolkit, cuDNN and TensorRT libraries by yourself.

NaeemKhan333, 2020-09-25:
I have installed all the given requirement as you said,but its giving me no detection your latest code.

NaeemKhan333, 2020-09-25:
Result Generated using your latest master branch code

[image]

NaeemKhan333, 2020-09-25:
@jkjung-avt it is giving result on the old commit although results are false, why it does not giving any result on the latest version of code. If installation could be the reason, it should not give result on the old commit of the code. Please guide me. Thanks

jkjung-avt, 2020-09-28:
I don't know what went wrong with your tests.  As I said previously, please focus on solving the problem with the latest code in this repo.

Do you see **any warnings/errors** when you:

1. build libyolo_layer.so (the plugin),
2. convert darknet model to onnx,
3. build onnx into TensorRT engine, and
4. run inference with the TensorRT engine?

NaeemKhan333, 2020-09-28:
# Output build the "yolo_layer" plugin

makeMakefile:20: "compute=75" is for GeForce RTX-2080 Ti.  Please make sure CUDA compute is set correctly for your system in the Makefile.
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -m64 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_75,code=compute_75 -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int, const nvinfer1::Dims *, int, const nvinfer1::DataType *, const nvinfer1::DataType *, const __nv_bool *, const __nv_bool *, nvinfer1::PluginFormat, int)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int, const nvinfer1::Dims *, int, const nvinfer1::DataType *, const nvinfer1::DataType *, const bool *, const bool *, nvinfer1::PluginFormat, int)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group

# Output for the yolo to ONNX

Parsing DarkNet cfg file...
Building ONNX graph...
graph yolov3-tiny-416 (
  %000_net[FLOAT, 1x3x416x416]
) initializers (
  %001_convolutional_bn_scale[FLOAT, 16]
  %001_convolutional_bn_bias[FLOAT, 16]
  %001_convolutional_bn_mean[FLOAT, 16]
  %001_convolutional_bn_var[FLOAT, 16]
  %001_convolutional_conv_weights[FLOAT, 16x3x3x3]
  %003_convolutional_bn_scale[FLOAT, 32]
  %003_convolutional_bn_bias[FLOAT, 32]
  %003_convolutional_bn_mean[FLOAT, 32]
  %003_convolutional_bn_var[FLOAT, 32]
  %003_convolutional_conv_weights[FLOAT, 32x16x3x3]
  %005_convolutional_bn_scale[FLOAT, 64]
  %005_convolutional_bn_bias[FLOAT, 64]
  %005_convolutional_bn_mean[FLOAT, 64]
  %005_convolutional_bn_var[FLOAT, 64]
  %005_convolutional_conv_weights[FLOAT, 64x32x3x3]
  %007_convolutional_bn_scale[FLOAT, 128]
  %007_convolutional_bn_bias[FLOAT, 128]
  %007_convolutional_bn_mean[FLOAT, 128]
  %007_convolutional_bn_var[FLOAT, 128]
  %007_convolutional_conv_weights[FLOAT, 128x64x3x3]
  %009_convolutional_bn_scale[FLOAT, 256]
  %009_convolutional_bn_bias[FLOAT, 256]
  %009_convolutional_bn_mean[FLOAT, 256]
  %009_convolutional_bn_var[FLOAT, 256]
  %009_convolutional_conv_weights[FLOAT, 256x128x3x3]
  %011_convolutional_bn_scale[FLOAT, 512]
  %011_convolutional_bn_bias[FLOAT, 512]
  %011_convolutional_bn_mean[FLOAT, 512]
  %011_convolutional_bn_var[FLOAT, 512]
  %011_convolutional_conv_weights[FLOAT, 512x256x3x3]
  %013_convolutional_bn_scale[FLOAT, 1024]
  %013_convolutional_bn_bias[FLOAT, 1024]
  %013_convolutional_bn_mean[FLOAT, 1024]
  %013_convolutional_bn_var[FLOAT, 1024]
  %013_convolutional_conv_weights[FLOAT, 1024x512x3x3]
  %014_convolutional_bn_scale[FLOAT, 256]
  %014_convolutional_bn_bias[FLOAT, 256]
  %014_convolutional_bn_mean[FLOAT, 256]
  %014_convolutional_bn_var[FLOAT, 256]
  %014_convolutional_conv_weights[FLOAT, 256x1024x1x1]
  %015_convolutional_bn_scale[FLOAT, 512]
  %015_convolutional_bn_bias[FLOAT, 512]
  %015_convolutional_bn_mean[FLOAT, 512]
  %015_convolutional_bn_var[FLOAT, 512]
  %015_convolutional_conv_weights[FLOAT, 512x256x3x3]
  %016_convolutional_conv_bias[FLOAT, 255]
  %016_convolutional_conv_weights[FLOAT, 255x512x1x1]
  %019_convolutional_bn_scale[FLOAT, 128]
  %019_convolutional_bn_bias[FLOAT, 128]
  %019_convolutional_bn_mean[FLOAT, 128]
  %019_convolutional_bn_var[FLOAT, 128]
  %019_convolutional_conv_weights[FLOAT, 128x256x1x1]
  %020_upsample_scale[FLOAT, 4]
[... 41 lines cut ...]
  %016_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%015_convolutional_lrelu, %016_convolutional_conv_weights, %016_convolutional_conv_bias)
  %019_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%014_convolutional_lrelu, %019_convolutional_conv_weights)
  %019_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%019_convolutional, %019_convolutional_bn_scale, %019_convolutional_bn_bias, %019_convolutional_bn_mean, %019_convolutional_bn_var)
  %019_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%019_convolutional_bn)
  %020_upsample = Upsample[mode = 'nearest'](%019_convolutional_lrelu, %020_upsample_scale)
  %021_route = Concat[axis = 1](%020_upsample, %009_convolutional_lrelu)
  %022_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [3, 3], strides = [1, 1]](%021_route, %022_convolutional_conv_weights)
  %022_convolutional_bn = BatchNormalization[epsilon = 9.99999974737875e-06, momentum = 0.990000009536743](%022_convolutional, %022_convolutional_bn_scale, %022_convolutional_bn_bias, %022_convolutional_bn_mean, %022_convolutional_bn_var)
  %022_convolutional_lrelu = LeakyRelu[alpha = 0.100000001490116](%022_convolutional_bn)
  %023_convolutional = Conv[auto_pad = 'SAME_LOWER', dilations = [1, 1], kernel_shape = [1, 1], strides = [1, 1]](%022_convolutional_lrelu, %023_convolutional_conv_weights, %023_convolutional_conv_bias)
  return %016_convolutional, %023_convolutional
}
Checking ONNX model...
Saving ONNX file...
Done.

#Output For the onnx_to_tensorrt

Loading ONNX file from path yolov3-tiny-416.onnx...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" to enable verbose logging.)
[TensorRT] WARNING: Half2 support requested on hardware without native FP16 support, performance will be negatively affected.
Completed creating engine.
Serialized the TensorRT engine to file: yolov3-tiny-416.trt

jkjung-avt, 2020-09-28:
The following warning is very strange.  You mentioned that you are using GTX-1080, which should support FP16 inferencing.  You should not see this warning.  Could you double-check that you are really using GTX-1080?  (Maybe you have mutiple GPUs in the system?)

> [TensorRT] WARNING: Half2 support requested on hardware without native FP16 support, performance will be negatively affected.

NaeemKhan333, 2020-09-28:
I am using an online GCP instance . The details are following

[image]

jkjung-avt, 2020-09-28:
So that's why!  You are using a Tesla K80, not GTX-1080.  Tesla K80 does not support F16.

You have to build FP32 TensorRT engines instead.  Please set `builder.fp16_mode` to `False` and rebuild the TensorRT engine.

https://github.com/jkjung-avt/tensorrt_demos/blob/1b274b874b59c0c7ed05f5efb2d89c0089fa3a15/yolo/onnx_to_tensorrt.py#L74

NaeemKhan333, 2020-09-28:
How I will build  FP32 TensorRT engines?

jkjung-avt, 2020-09-28:
Modify the source code as I've told you, then execute the same commands again.

NaeemKhan333, 2020-09-28:
Ok Thanks for such a great Help

NaeemKhan333, 2020-09-28:
@jkjung-avt By doing above modifications in source, warning gets removed but the output results are same.

jkjung-avt, 2020-09-28:
Referring to https://developer.nvidia.com/cuda-gpus, you could see that "compute capability" of Tesla K80 is "3.7".  It's a rather old GPU.  According to [TensorRT documentation](https://docs.nvidia.com/deeplearning/tensorrt/support-matrix/index.html#hardware-precision-matrix):

> Note: Support for CUDA Compute Capability version 3.0 has been removed. Support for CUDA Compute Capability versions below 5.0 may be removed in a future release and is now deprecated.

I would suggest you to test the code with a more modern GPU.

jkjung-avt, 2020-10-05:
https://github.com/jkjung-avt/tensorrt_demos/issues/240#issuecomment-698743480

@NaeemKhan333  Are you able to run the code successfully on your GTX-1080?  Can we close this issue?

jkjung-avt, 2020-10-11:
Close due to no response.

closed 2020-10-11 by jkjung-avt