jkjung-avt/tensorrt_demos #243: #assertiongridAnchorPlugin.cpp,205 Aborted (core dumped)

jishang1602, 2020-09-30:
I want to get the object detection once the images are going to be dumped into the older. So on any new images i want to do object detection without loading the model every time.My code is below . getting an error of #assertiongridAnchorPlugin.cpp,205 Aborted (core dumped)

import time
import argparse
import os
import cv2
#from utils.mtcnn import TrtMtcnn
import pycuda.autoinit  # This is needed for initializing CUDA driver
from utils.ssd_classes import get_cls_dict
from utils.ssd import TrtSSD
#from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

from watchdog.observers import Observer
from watchdog.events import FileSystemEventHandler

directory = "/home/jishang/out-test/"
filename_prefix ="out"

WINDOW_NAME = 'TrtSsdDemoAsync'
MAIN_THREAD_TIMEOUT = 20.0  # 20 seconds
INPUT_HW = (300, 300)
SUPPORTED_MODELS = [
    'ssd_mobilenet_v1_coco',
    'ssd_mobilenet_v1_egohands',
    'ssd_mobilenet_v2_coco',
    'ssd_mobilenet_v2_egohands',
    'ssd_inception_v2_coco',
    'ssdlite_mobilenet_v2_coco',
]

def add_camera_args(parser):
    """Add parser augument for camera options."""
    parser.add_argument('--image', type=str, default=None,
                        help='image file name, e.g. dog.jpg')
    parser.add_argument('--video', type=str, default=None,
                        help='video file name, e.g. traffic.mp4')
    parser.add_argument('--video_looping', action='store_true',
                        help='loop around the video file [False]')
    parser.add_argument('--rtsp', type=str, default=None,
                        help=('RTSP H.264 stream, e.g. '
                              'rtsp://admin:123456@192.168.1.64:554'))
    parser.add_argument('--rtsp_latency', type=int, default=200,
                        help='RTSP latency in ms [200]')
    parser.add_argument('--usb', type=int, default=None,
                        help='USB webcam device id (/dev/video?) [None]')
    parser.add_argument('--onboard', type=int, default=None,
                        help='Jetson onboard camera [None]')
    parser.add_argument('--copy_frame', action='store_true',
                        help=('copy video frame internally [False]'))
    parser.add_argument('--do_resize', action='store_true',
                        help=('resize image/video [False]'))
    parser.add_argument('--width', type=int, default=640,
                        help='image width [640]')
    parser.add_argument('--height', type=int, default=480,
                        help='image height [480]')
    return parser

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time face detection with TrtMtcnn on Jetson '
            'Nano')
    parser = argparse.ArgumentParser(description=desc)
    parser = add_camera_args(parser)
    parser.add_argument('--minsize', type=int, default=40,
                        help='minsize (in pixels) for detection [40]')
    parser.add_argument('-m', '--model', type=str,
                        default='ssd_mobilenet_v1_coco',
                        choices=SUPPORTED_MODELS)
    args = parser.parse_args()
    return args

args = parse_args()
image = args.rtsp

cls_dict = get_cls_dict(args.model.split('_')[-1])
trt_ssd = TrtSSD(args.model, INPUT_HW)
vis = BBoxVisualization(cls_dict)
class Watcher:
	DIRECTORY_TO_WATCH = "/home/jishang/test/"
	def __init__(self):
		self.observer = Observer()
	def run(self):
		event_handler = Handler()
		self.observer.schedule(event_handler, self.DIRECTORY_TO_WATCH, recursive=True)
		self.observer.start()
		try:
			while True:
				time.sleep(0.001)
		except:
			self.observer.stop()
		print ("Error")
		self.observer.join()

class Handler(FileSystemEventHandler):
	@staticmethod
	def on_any_event(event):
		if event.is_directory:
			return None
		elif event.event_type == 'created':
			# Take any action here when a file is first created.
			img_jpg=event.src_path
			print(img_jpg)
			#img=cv2.imread(img_jpg)
			#if img is not None:
			#	print("No image to load")
			#cv2.imshow('image', img) 
			#print ("Received created event - " ,event.src_path)
		elif event.event_type == 'modified':
			# Taken any action here when a file is modified.
			#print ("Received modified event -" , event.src_path)
			#time.sleep(1)
			tic = time.time()
			img_jpg=event.src_path
			print(img_jpg)
			img=cv2.imread(img_jpg)
			detect(img, trt_ssd, conf_th=0.3, vis=vis)
			toc =time.time()
			print("WatchDog time in milli", toc - tic)

def detect(img,trt_ssd, conf_th, vis):
	"""Continuously capture images from camera and do object detection."""
	full_scrn = False
	fps = 0.0
	tic = time.time()
	boxes, confs, clss = trt_ssd.detect(img, 0.4)
	img = vis.draw_bboxes(img, boxes, confs, clss)
	
	
	toc = time.time() 
	curr_fps = 1.0 / (toc - tic)
	print("Detect Loop time in milli ",toc-tic)
	filename = str(filename_prefix) +str(toc) + ".jpg"
	cv2.imwrite(os.path.join(directory , filename), img)
	

def main():
	#args = parse_args()
	#mtcnn = TrtMtcnn()
	#image = args.rtsp
	if not image:
		#wait for file creation and process it
		print(" In watch")
		w = Watcher()
		w.run()
	else:
		#contineous process on rtsp stream
		cam=cv2.VideoCapture(image)
		if not cam.isOpened():
			raise SystemExit('ERROR: failed to open camera!')
if __name__ == '__main__':
	main()

 python3 img_ss.py 
 In watch
/home/jishang/test/1.jpg
/home/jishang/test/1.jpg
#assertiongridAnchorPlugin.cpp,205
Aborted (core dumped)

jishang1602, 2020-09-30:
`import time
import argparse
import os
import cv2
#from utils.mtcnn import TrtMtcnn
import pycuda.autoinit  # This is needed for initializing CUDA driver
from utils.ssd_classes import get_cls_dict
from utils.ssd import TrtSSD
#from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

from watchdog.observers import Observer
from watchdog.events import FileSystemEventHandler

directory = "/home/cloudvms/out-test/"
filename_prefix ="out"

WINDOW_NAME = 'TrtSsdDemoAsync'
MAIN_THREAD_TIMEOUT = 20.0  # 20 seconds
INPUT_HW = (300, 300)
SUPPORTED_MODELS = [
    'ssd_mobilenet_v1_coco',
    'ssd_mobilenet_v1_egohands',
    'ssd_mobilenet_v2_coco',
    'ssd_mobilenet_v2_egohands',
    'ssd_inception_v2_coco',
    'ssdlite_mobilenet_v2_coco',
]

def add_camera_args(parser):
    """Add parser augument for camera options."""
    parser.add_argument('--image', type=str, default=None,
                        help='image file name, e.g. dog.jpg')
    parser.add_argument('--video', type=str, default=None,
                        help='video file name, e.g. traffic.mp4')
    parser.add_argument('--video_looping', action='store_true',
                        help='loop around the video file [False]')
    parser.add_argument('--rtsp', type=str, default=None,
                        help=('RTSP H.264 stream, e.g. '
                              'rtsp://admin:123456@192.168.1.64:554'))
    parser.add_argument('--rtsp_latency', type=int, default=200,
                        help='RTSP latency in ms [200]')
    parser.add_argument('--usb', type=int, default=None,
                        help='USB webcam device id (/dev/video?) [None]')
    parser.add_argument('--onboard', type=int, default=None,
                        help='Jetson onboard camera [None]')
    parser.add_argument('--copy_frame', action='store_true',
                        help=('copy video frame internally [False]'))
    parser.add_argument('--do_resize', action='store_true',
                        help=('resize image/video [False]'))
    parser.add_argument('--width', type=int, default=640,
                        help='image width [640]')
    parser.add_argument('--height', type=int, default=480,
                        help='image height [480]')
    return parser

def parse_args():
    """Parse input arguments."""
    desc = ('Capture and display live camera video, while doing '
            'real-time face detection with TrtMtcnn on Jetson '
            'Nano')
    parser = argparse.ArgumentParser(description=desc)
    parser = add_camera_args(parser)
    parser.add_argument('--minsize', type=int, default=40,
                        help='minsize (in pixels) for detection [40]')
    parser.add_argument('-m', '--model', type=str,
                        default='ssd_mobilenet_v1_coco',
                        choices=SUPPORTED_MODELS)
    args = parser.parse_args()
    return args

args = parse_args()
image = args.rtsp

cls_dict = get_cls_dict(args.model.split('_')[-1])
trt_ssd = TrtSSD(args.model, INPUT_HW)
vis = BBoxVisualization(cls_dict)
class Watcher:
	DIRECTORY_TO_WATCH = "/home/cloudvms/test/"
	def __init__(self):
		self.observer = Observer()
	def run(self):
		event_handler = Handler()
		self.observer.schedule(event_handler, self.DIRECTORY_TO_WATCH, recursive=True)
		self.observer.start()
		try:
			while True:
				time.sleep(0.001)
		except:
			self.observer.stop()
		print ("Error")
		self.observer.join()

class Handler(FileSystemEventHandler):
	@staticmethod
	def on_any_event(event):
		if event.is_directory:
			return None
		elif event.event_type == 'created':
			# Take any action here when a file is first created.
			img_jpg=event.src_path
			print(img_jpg)
			#img=cv2.imread(img_jpg)
			#if img is not None:
			#	print("No image to load")
			#cv2.imshow('image', img) 
			#print ("Received created event - " ,event.src_path)
		elif event.event_type == 'modified':
			# Taken any action here when a file is modified.
			#print ("Received modified event -" , event.src_path)
			#time.sleep(1)
			tic = time.time()
			img_jpg=event.src_path
			print(img_jpg)
			img=cv2.imread(img_jpg)
			detect(img, trt_ssd, conf_th=0.3, vis=vis)
			toc =time.time()
			print("WatchDog time in milli", toc - tic)

import pycuda.autoinit  # This is needed for initializing CUDA driver
from utils.ssd_classes import get_cls_dict
from utils.ssd import TrtSSD
from utils.camera import add_camera_args, Camera
from utils.display import open_window, set_display, show_fps
from utils.visualization import BBoxVisualization

def detect(img,trt_ssd, conf_th, vis):
	"""Continuously capture images from camera and do face detection."""
	full_scrn = False
	fps = 0.0
	tic = time.time()
	boxes, confs, clss = trt_ssd.detect(img, 0.4)
	img = vis.draw_bboxes(img, boxes, confs, clss)
	
	#dets, landmarks = mtcnn.detect(img, minsize=minsize)
	#print('{} face(s) found'.format(len(dets)))
	#img = show_faces(img, dets, landmarks)
	toc = time.time() 
	curr_fps = 1.0 / (toc - tic)
	print("Detect Loop time in milli ",toc-tic)
	filename = str(filename_prefix) +str(toc) + ".jpg"
	cv2.imwrite(os.path.join(directory , filename), img)
	

def main():
	#args = parse_args()
	#mtcnn = TrtMtcnn()
	#image = args.rtsp
	if not image:
		#wait for file creation and process it
		print(" In watch")
		w = Watcher()
		w.run()
	else:
		#contineous process on rtsp stream
		cam=cv2.VideoCapture(image)
		if not cam.isOpened():
			raise SystemExit('ERROR: failed to open camera!')
if __name__ == '__main__':
	main()`

jkjung-avt, 2020-09-30:
I think your event_handler() is executed from a child thread.  So you could try to replace this line:

```
trt_ssd = TrtSSD(args.model, INPUT_HW)
```

with:

```
trt_ssd = TrtSSD(args.model, INPUT_HW, cuda_ctx=pycuda.autoinit.context)
```

Check if that solves the problem for you.

jishang1602, 2020-09-30:
yeah. this works. Thank you for the quick response

closed 2020-09-30 by jishang1602