jkjung-avt/tensorrt_demos #244: move trt_mtcnn.py to parent directory

AnasMK, 2020-10-01:
Thanks for this great repo. I'm trying to use tensorrt_demos as library so I moved the `trt_mtcnn.py` to the parent directory, and fix some imports but it doesn't work give me error. Could you please assist me with this.

These the changes I have done
1. moved  `trt_mtcnn.py` to parent directory | `project/tensorrt_demos/trt_mtcnn.py =>  project/trt_mtcnn.py`
2. changed the utils package imports in `trt_mtcnn.py` to :
```python
from tensorrt_demos.utils.camera import add_camera_args, Camera
from tensorrt_demos.utils.display import open_window, set_display, show_fps
from tensorrt_demos.utils.mtcnn import TrtMtcnn
```
3. changed pytrt import in `tensorrt_demos/utils/mtcnn.py`  to `import tensorrt_demos.pytrt as pytrt`
After this I tired to run `project/trt_mtcnn.py` but gave me this error 
```shell
Parameter check failed at: runtime.cpp::deserializeCudaEngine::30, condition: (blob) != nullptr
Assertion fail in file 'trtNet.cpp' line 146: _engine is null
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted (core dumped)
```
I don't know C++ so I couldn't figure out what  need to be changed.

jkjung-avt, 2020-10-01:
The MTCNN inference code looks for TensorRT engine files at 'mtcnn/det1.engine', 'mtcnn/det2.engine', and 'mtcnn/det3.engine'.  You need to copy the engine files to those locations (or modify the following 3 lines of code). 

https://github.com/jkjung-avt/tensorrt_demos/blob/1b274b874b59c0c7ed05f5efb2d89c0089fa3a15/utils/mtcnn.py#L439-L441

AnasMK, 2020-10-04:
Thank you it is working now

closed 2020-10-04 by AnasMK