jkjung-avt/tensorrt_demos #245: Applying TLT generated trt-engines

ZeljkoKovacevic, 2020-10-02:
I'm trying to run inference with trt-engines that have been trained, exported and converted with the NVIDIA TLT framework in order to run on Jetson Xavier NX.
Therefore I modified your ssd demo code slightly (we're using TLTs ssd_resnet50).
Basically I had to change the shape (we're using wxh 768x432) and adjust the image preprocessing.

Model is okay as we tested on NVIDIAs docker container and inference is okay there.

Code is running on the NX with no errors, however, I don't get any valid detections...if any.
- TensorRT open source installed as demanded by docu (needed for plugins)
- I tried several image preprocessing strategies as the major frameworks use different input formats  (tf, torch, caffe format)...yet no luck.

Do you have a hint what might be going on?

jkjung-avt, 2020-10-02:
I'm not familiar with TLT, so I'm not able to give you a definitive answer.

But you are looking in the right direction, I think the TLT trained model would require "preproessing" and "postprocessing" that are different from my trt_ssd implementation.  You just need to figure those out and modify the code accordingly.  For example,

* For preprocessing: In addition to resizing the image, you should also pay attention to, say, ordering of the color channels (RGB or BGR), ordering the channel dimension (NCHW or NHWC), mean subtraction and scale factor (1/127.5 or 1/255), etc.

* For postprocessing: You need to know the format of the output data.  And you might need to do NMS if that's not done in the TLT model.

ZeljkoKovacevic, 2020-10-10:
Thanks a lot for your hints.
Indeed there where multiple issues
1. convert to tensort rt engine with NCHW format
2. TLT scales the image to the tensorrt engine shape keeping the aspect ratio, hence there is either a blank region at the bottom or at the right side
3. prediction output coordinates a normalized between 0 and 1

Thanks again..your repos are invaluable.

closed 2020-10-10 by ZeljkoKovacevic