jkjung-avt/tensorrt_demos #246: Issue while converting from ONNX to TensorRT Engine

Darshcg, 2020-10-02:
Hi,
I am Converting my Custom Pytorch Model to TRT Engine file. I am able to convert the model to Onnx, but I am facing an Issue while converting from Onnx to the Engine file. 
I am getting the error as:
AttributeError: ‘NoneType’ object has no attribute ‘serialize’.
I am giving the correct shape as an input to the build engine function.
And for the Resnet50 model, I am able to convert successfully but for my custom model, I am facing the above issue.
Please suggest if there is any specific reason behind that.
[image]

Thanks

Darshcg, 2020-10-02:
I tried with trtexec command, I am getting the error as attached below:

While parsing node number 195 [Resize]:
ERROR: ModelImporter.cpp:124 In function parseGraph:
[5] Assertion failed: ctx->tensors().count(inputName)
[09/25/2020-11:02:52] [E] Failed to parse onnx file
[09/25/2020-11:02:52] [E] Parsing model failed
[09/25/2020-11:02:52] [E] Engine creation failed
[09/25/2020-11:02:52] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec # trtexec --onnx=last.onnx --batch=1 --saveEngine=last.engine

The command I used: trtexec --onnx=last.onnx --batch=1 --saveEngine=last.engine

jkjung-avt, 2020-10-02:
I think this issue is not directly related code in this repo...  I probably cannot help you.

jkjung-avt, 2020-10-02:
https://github.com/NVIDIA/TensorRT/issues/812#issuecomment-702757860

closed 2020-10-06 by jkjung-avt