jkjung-avt/tensorrt_demos #248: Inference code for the custom TRT Model

Darshcg, 2020-10-06:
Hi,

What are the changes to be made in this(https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_ssd.py and https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_yolo.py) in order to work on Custom TRT model inference.

Thanks

jkjung-avt, 2020-10-06:
If you are using custom trained ssd_mobilenet_v1, ssd_mobilenet_v2 or ssd_inception_v2 models, refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/76

If you are using custom trained yolov3/yolov4 models, refer to: [TensorRT YOLOv3 For Custom Trained Models](https://jkjung-avt.github.io/trt-yolov3-custom/)

If your custom model differs a lot from those standard models, you might need to modify the source code quite a bit.  For example, if your model uses some layers not directly supported by TensorRT, you might need to program a plugin by yourself.

closed 2020-10-11 by jkjung-avt

Darshcg, 2020-10-12:
Hi @jkjung-avt,

Thanks a lot for your guidance.