jkjung-avt/tensorrt_demos #250: Question about custom plugin and precision, anyone can tell me your answer.

jch-wang, 2020-10-08:
1. I only see the mish activation function in yolo_to_onnx.py. Doesn't it need to be implemented with plugin? 
2.If I need to try with different precision (fp32, fp16, int8), where should I set it?

jkjung-avt, 2020-10-08:
> 1. I only see the mish activation function in yolo_to_onnx.py. Doesn't it need to be implemented with plugin?

Please refer to [this discussion](https://forums.developer.nvidia.com/t/cannot-parse-custom-layer-by-onnx-tensorrt/141902/8?u=jkjung13).  I think my implementation (replacing "Mish" with "Softplus” + “Tanh” + “Mul”) is correct.  Using a plugin might improve inference speed, but should produce the same detection resutls.

> 2. If I need to try with different precision (fp32, fp16, int8), where should I set it?

By default, my "onnx_to_yolo.py" builds TensorRT engine in FP16 mode.  If you'e like to build INT8 engine, just add "--int8" in the command line.  (Refer to README.md.)

For FP32, you need to disable FP16 setting in the source code.  I've highlighted the relevant code below, for TensorRT 6 and 7, respectively.

https://github.com/jkjung-avt/tensorrt_demos/blob/2c37329c4aae9409d1db907a77eb97d581c75610/yolo/onnx_to_tensorrt.py#L122

https://github.com/jkjung-avt/tensorrt_demos/blob/2c37329c4aae9409d1db907a77eb97d581c75610/yolo/onnx_to_tensorrt.py#L134

jch-wang, 2020-10-08:
> config.set_flag(trt.BuilderFlag.FP16)

Thank you for your quick answer. I think I have got the answer I want. I'll try it on my jedson platform.

jch-wang, 2020-10-09:
> > 1. I only see the mish activation function in yolo_to_onnx.py. Doesn't it need to be implemented with plugin?
> 
> Please refer to [this discussion](https://forums.developer.nvidia.com/t/cannot-parse-custom-layer-by-onnx-tensorrt/141902/8?u=jkjung13). I think my implementation (replacing "Mish" with "Softplus” + “Tanh” + “Mul”) is correct. Using a plugin might improve inference speed, but should produce the same detection resutls.
> 
> > 1. If I need to try with different precision (fp32, fp16, int8), where should I set it?
> 
> By default, my "onnx_to_tensorrt.py" builds TensorRT engine in FP16 mode. If you'e like to build INT8 engine, just add "--int8" in the command line. (Refer to README.md.)
> 
> For FP32, you need to disable FP16 setting in the source code. I've highlighted the relevant code below, for TensorRT 6 and 7, respectively.
> 
> https://github.com/jkjung-avt/tensorrt_demos/blob/2c37329c4aae9409d1db907a77eb97d581c75610/yolo/onnx_to_tensorrt.py#L122
> 
> https://github.com/jkjung-avt/tensorrt_demos/blob/2c37329c4aae9409d1db907a77eb97d581c75610/yolo/onnx_to_tensorrt.py#L134

I wanted to generate an int8 precision yolov3-416 engine file, but the following error occurred:
```
lqs@lqs-desktop:~/tensorrt_demos_yolov3_4/yolo$ python3 onnx_to_tensorrt.py -m yolov3-416 --int8
Loading the ONNX file...
Adding yolo_layer plugins...
Building an engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 195, in <module>
    main()
  File "onnx_to_tensorrt.py", line 184, in main
    args.model, args.category_num, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 147, in build_engine
    'calib_%s.bin' % model_name)
  File "/home/lqs/tensorrt_demos_yolov3_4/yolo/calibrator.py", line 97, in __init__
    raise FileNotFoundError('%s does not exist' % img_dir)
FileNotFoundError: calib_images does not exist
Exception ignored in: <bound method YOLOEntropyCalibrator.__del__ of <calibrator.YOLOEntropyCalibrator object at 0x7f4e1030f8>>
Traceback (most recent call last):
  File "/home/lqs/tensorrt_demos_yolov3_4/yolo/calibrator.py", line 120, in __del__
    del self.device_input  # free CUDA memory
AttributeError: device_input
```
Similarly, when I run build_ int8_ engines.sh File, a similar error report appears。And I didn't see INT8 calibration cache files about yolovX-416 under tensorrt_demos/yolo/calib_cache。How can I solve this mistake？

jch-wang, 2020-10-09:
> Please refer to [this discussion](https://forums.developer.nvidia.com/t/cannot-parse-custom-layer-by-onnx-tensorrt/141902/8?u=jkjung13). I think my implementation (replacing "Mish" with "Softplus” + [...]

You mean for TensorRT 7, if I want to use FP32, I need to comment out "config.set_flag(trt.BuilderFlag.FP16)" in file onnx_to_tensorrt.py? 
I'm sorry, but I have a lot of questions. I hope it doesn't bother you.

jkjung-avt, 2020-10-09:
> I wanted to generate an int8 precision yolov3-416 engine file, but the following error occurred:

Did you read and follow step #1 of [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core)?  You need to create the "calib_images" directory and put some images for calibration in it.

> You mean for TensorRT 7, if I want to use FP32, I need to comment out "config.set_flag(trt.BuilderFlag.FP16)" in file onnx_to_tensorrt.py?

Yes

jch-wang, 2020-10-11:
> Did you read and follow step #1 of [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core)? You need to create the "calib_images" directory [...]

By default, your "build_engine.py" builds SSD TensorRT engine in FP16 mode. If I want to build INT8/FP32 engine , What should I do? It seems different from yolo's，I only see the following related commands in "build_engine.py"：
`builder.fp16_mode = True`

jkjung-avt, 2020-10-11:
> By default, your "build_engine.py" builds SSD TensorRT engine in FP16 mode. If I want to build INT8/FP32 engine , What should I do? It seems different from yolo's.  I only see the following related commands in "build_engine.py"：
`builder.fp16_mode = True`

For FP32, you could just modify the source code to `builder.fp16_mode = False`.

For INT8, you could reference my TensorRT YOLO implementation and modify the code by yourself.  You could reuse most of my [calibrator.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/yolo/calibrator.py) code, so the required modification is not much.

I myself don't have a plan to implement INT8 for TensorRT UFF SSD.  I think it's a good exercise for people who'd like to learn TensorRT's python API for INT8 optimization.

closed 2020-10-11 by jch-wang