jkjung-avt/tensorrt_demos #251: I have a question.

hjlee9182, 2020-10-08:
[image]

Modify some code in 'eval_yolo.py'.

But that result FPS is lower than 'README.md' FPS. ( about 1~5FPS)

So I don't know what is problem.
Is there something wrong?

jkjung-avt, 2020-10-08:
I measure FPS with "trt_yolo.py" directly.

You would see a lower FPS with "eval_yolo.py" because:

1. Detection confidence threshold (1e-2) is lower.  So the postprocessing (NMS) would take longer.
2. Reading various jpg files causes additional loading on the CPU and I/O.  It would slow down inference code a little bit.

hjlee9182, 2020-10-08:
ok thank u

closed 2020-10-08 by hjlee9182