jkjung-avt/tensorrt_demos #252: RTSP stream detection failed

jch-wang, 2020-10-08:
After I generate the engine file, I want to detect RTSP. The RTSP stream is available, but I get the following error message。Can I have some suggestions？
`lqs@lqs-desktop:~/tensorrt_demos_yolov3_4$ python3 trt_yolo.py --rtsp rtsp://admin:a12345678@192.168.1.67 -m yolov4-416

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_caps_is_empty: assertion 'GST_IS_CAPS (caps)' failed

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_caps_truncate: assertion 'GST_IS_CAPS (caps)' failed

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_caps_fixate: assertion 'GST_IS_CAPS (caps)' failed

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_structure_get_string: assertion 'structure != NULL' failed

(python3:10138): GStreamer-CRITICAL **: 13:38:51.201: gst_mini_object_unref: assertion 'mini_object != NULL' failed
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Allocating new output: 1920x1088 (x 11), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1920, nFrameHeight = 1080 
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (896) open OpenCV | GStreamer warning: unable to query duration of stream
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=1, value=0, duration=-1
Traceback (most recent call last):
  File "/home/lqs/tensorrt_demos_yolov3_4/utils/yolo_with_plugins.py", line 257, in __init__
    allocate_buffers(self.engine, grid_sizes)
  File "/home/lqs/tensorrt_demos_yolov3_4/utils/yolo_with_plugins.py", line 153, in allocate_buffers
    stream = cuda.Stream()
AttributeError: module 'pycuda.driver' has no attribute 'Stream'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "trt_yolo.py", line 116, in <module>
    main()
  File "trt_yolo.py", line 103, in main
    trt_yolo = TrtYOLO(args.model, (h, w), args.category_num)
  File "/home/lqs/tensorrt_demos_yolov3_4/utils/yolo_with_plugins.py", line 259, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources`

jkjung-avt, 2020-10-08:
`AttributeError: module 'pycuda.driver' has no attribute 'Stream'`

This indicates you might have problem with your "pycuda" installation...

jch-wang, 2020-10-08:
> `AttributeError: module 'pycuda.driver' has no attribute 'Stream'`
> 
> This indicates you might have problem with your "pycuda" installation...

It worked after I installed it again。
>>Under folder ssd，I see the “trt_version=5” and “trt_version=6 ” in the file"install.sh". Can I use "Demo3 SSD" in "trt_version=7"?

jkjung-avt, 2020-10-08:
> Under folder ssd，I see the “trt_version=5” and “trt_version=6 ” in the file"install.sh". Can I use "Demo3 SSD" in "trt_version=7"?

Yes.  The code works for TensorRT 7.x.

The "FlattenConcat" plugin is built-in in TensorRT 7+.  That's why I don't need to included a pre-compiled .so file of it for TensorRT 7+ anymore.

jch-wang, 2020-10-08:
> Yes. The code works for TensorRT 7.x.
>
> The "FlattenConcat" plugin is built-in in TensorRT 7+. That's why I don't need to included a pre-compiled .so file of it for TensorRT 7+ anymore.

Great！ 
You mean when TensorRT version is 7, I don't need to execute the command line "$/ install.sh "？
Can I just use the following command to build the engine？：
```
$ cd  $ {HOME} / project / tensorrt_demos / ssd
$ ./build_engines.sh
```

jkjung-avt, 2020-10-08:
You still need to run the "install.sh" script in the "ssd" directory.  It will install "pycuda" and patch "graphsurgeon" for you.  The script will not make the symbolic link to the .so file if TensorRT version is not 5 or 6.  Refer to the code below:

https://github.com/jkjung-avt/tensorrt_demos/blob/3ec5d15479515669552efb0e7285b4e8ee317def/ssd/install.sh#L32-L34

jch-wang, 2020-10-08:
> You still need to run the "install.sh" script in the "ssd" directory. It will install "pycuda" and patch "graphsurgeon" for you. The script will not make the symbolic link to the .so file if [...]

Oh, I see. Your quick reply has made my progress very smooth. Thank you again.

closed 2020-10-08 by jch-wang