jkjung-avt/tensorrt_demos #253: onnx_to_tensorrt.py takes forever

xpeedster, 2020-10-08:
Hi, 

I am using a Jetson Nano Developer Kit (4GB, not de new 2GB one) and I followed your guide at [Jetpack 4.4 for Jetson Nano](https://jkjung-avt.github.io/jetpack-4.4/) Everything worked fine untill the last step on the Yolo Demo where after doing:
`python3 onnx_to_tensorrt.py -m yolov3-tiny-416`
This takes forever.

I mean, no errors shown, yet. However, you mention it takes just a bit longer than half an hour, but it has been 1 hour and a half and it keeps running.

Do you have any idea on what the problem can be?

Thanks in advance for your help.

jkjung-avt, 2020-10-08:
Please enable "-v" (verbose) so that you could see the progress.

```
$ python3 onnx_to_tensorrt.py -v -m yolov3-tiny-416
```

In addition, executing `sudo nvpmodel -m 0; sudo jetson_clocks` to maxmize CPU clock would help quite a bit.

xpeedster, 2020-10-08:
Okay, I will enable verbose.

I got MAXN model active, but not jetson clocks. I will do so.

Will report any news on this. Thanks for the help.

xpeedster, 2020-10-08:
Well, done it again, using verbose. It worked this time. I'm closing the issue.

Thanks for your help and your work.

closed 2020-10-08 by xpeedster