jkjung-avt/tensorrt_demos #256: FPS with great difference in the same environment，Why  -_-?

jch-wang, 2020-10-09:
Have you ever used the example of NVIDIA in  [https://github.com/dusty-nv/jetson-inference](url)？
On my Jetson Xavier NX, I tested the ssd-mobilenet-v2-coco contained in it, which can reach about 150 FPS. 
In the same environment, I tested ssd-mobilenet-v2-coco in your code, but it can only reach 20 FPS. 
Can you give me some advice? I'm confused right now.

jkjung-avt, 2020-10-09:
That's strange.  I got 22.7 FPS on **Jetson Nano**.  I will pay attention to this if I have the chance to test it on Jetson Xavier NX.

Otherwise, I think my posts [here](https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-663593512) and [here](https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-665105334) help to clarify a few things.  Please take a look.

jkjung-avt, 2020-10-10:
There's another thing.  **The way I calculate FPS** in "trt_ssd_async.py", "trt_ssd.py" and "trt_yolo.py" is to **include image preprocessing, inference, postprocessing, bounding-box drawing and image display**.  NVIDIA usually refers to FPS by only considering how much time TensorRT inferencing takes.  FPS numbers could differ quite a lot between these 2 approaches especially when the model is relatively small and runs fast.

jch-wang, 2020-10-10:
> There's another thing. **The way I calculate FPS** in "trt_ssd_async.py", "trt_ssd.py" and "trt_yolo.py" is to **include image preprocessing, inference, postprocessing, bounding-box drawing and [...]

Do you mean that in [https://github.com/dusty-nv/jetson-inference](url), only the time spent in TensorRT inferencing is used to calculate FPS？I want to confirm this answer again。I didn't read the specific code in jetson-inference because I didn't know much about C andC++.

jkjung-avt, 2020-10-10:
I didn't read jetson-inference source code either.  So I don't know.

jch-wang, 2020-10-11:
> I didn't read jetson-inference source code either. So I don't know.
Dusty-nv told me in his jetson-inference，What is displayed in the status bar is the DNN processing time.You can have a look [here](https://github.com/dusty-nv/jetson-inference/issues/754)

closed 2020-10-12 by jch-wang