jkjung-avt/tensorrt_demos #257: mAP evaluation of FP32 and FP16

jch-wang, 2020-10-12:
I evaluated yolov4-416 and yolov4-608 with different precision on my Jetson Xaiver NX, and the result was that fp32 was not better than fp16 in 12 evaluation indexes. Most indicators of fp32 are equal to fp16, and there are 2 to 3 indicators worse than fp16. For example:
```
YOLOV4-608-FP16
Average Precision  (AP) @[ IoU=0.50:0.95 | area=   all | maxDets=100 ] = 0.488
 Average Precision  (AP) @[ IoU=0.50     | area=   all | maxDets=100 ] = 0.736
 Average Precision  (AP) @[ IoU=0.75     | area=   all | maxDets=100 ] = 0.533
 Average Precision  (AP) @[ IoU=0.50:0.95 | area= small | maxDets=100 ] = 0.327
 Average Precision  (AP) @[ IoU=0.50:0.95 | area=medium | maxDets=100 ] = 0.548
 Average Precision  (AP) @[ IoU=0.50:0.95 | area= large | maxDets=100 ] = 0.604
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets=  1 ] = 0.354
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets= 10 ] = 0.560
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets=100 ] = 0.586
 Average Recall     (AR) @[ IoU=0.50:0.95 | area= small | maxDets=100 ] = 0.411
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=medium | maxDets=100 ] = 0.646
 Average Recall     (AR) @[ IoU=0.50:0.95 | area= large | maxDets=100 ] = 0.723
```
```
YOLOV4-608-FP32
Average Precision  (AP) @[ IoU=0.50:0.95 | area=   all | maxDets=100 ] = 0.488
 Average Precision  (AP) @[ IoU=0.50      | area=   all | maxDets=100 ] = 0.736
 Average Precision  (AP) @[ IoU=0.75      | area=   all | maxDets=100 ] = 0.533
 Average Precision  (AP) @[ IoU=0.50:0.95 | area= small | maxDets=100 ] = 0.327
 Average Precision  (AP) @[ IoU=0.50:0.95 | area=medium | maxDets=100 ] = 0.548
 Average Precision  (AP) @[ IoU=0.50:0.95 | area= large | maxDets=100 ] = 0.604
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets=  1 ] = 0.354
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets= 10 ] = 0.560
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=   all | maxDets=100 ] = 0.586
 Average Recall     (AR) @[ IoU=0.50:0.95 | area= small | maxDets=100 ] = 0.411
 Average Recall     (AR) @[ IoU=0.50:0.95 | area=medium | maxDets=100 ] = 0.646
 Average Recall     (AR) @[ IoU=0.50:0.95 | area= large | maxDets=100 ] = 0.722
```
They were all evaluated on dataset COCO-val017。A similar situation has occurred on yolov3。Is this not true? Shouldn't fp32 be better than fp16？

jkjung-avt, 2020-10-12:
This looks normal.  FP16 or INT8 optimized models might, in some cases, actually generalize slightly better than the original FP32 model.

jch-wang, 2020-10-12:
> This looks normal. FP16 or INT8 optimized models might, in some cases, actually generalize slightly better than the original FP32 model.

In  "Demo #6: Using INT8 and DLA core ",you tested the yolov4-int8-608 models on your Jetson Xavier NX DevKit with JetPack-4.4 (TensorRT 7.1.3.4)，and It reaches 20.5 FPS。but I tested the yolov4-int8-608 models on my Jetson Xavier NX can only reach 10.4FPS.I choose to  the  "calibration picture"  is randomly deleted after leaving 1000, the following code is not used，is this the reason?
```
$ for jpg in $(ls -1 ${HOME}/data/coco/images/val2017/*.jpg | sort -R | head -1000); do \
    cp ${HOME}/data/coco/images/val2017/${jpg} calib_images/; \
  done
```

jkjung-avt, 2020-10-12:
> I choose to the "calibration picture" is randomly deleted after leaving 1000, the following code is not used.  Is this the reason?

No, you don't need to use the same command.  You could pick the images for calibration whatever way you want.  Just make sure they are diverse enough and representative to what you'd be inferencing with.

The calibration images are used to create calibration cache during the process of building INT8 engines.  They are not needed during inferencing either.

jch-wang, 2020-10-13:
> No, you don't need to use the same command. You could pick the images for calibration whatever way you want. Just make sure they are diverse enough and representative to what you'd be inferencing [...]

So why is the FPS I tested only half of yours? What are the possible causes? I didn't change the code. I'd like to get some advice from you. Maybe I've overlooked some information.

jkjung-avt, 2020-10-13:
I have no idea.  If you start with a clean JetPack-4.4 installation of Jetson Xavier NX and follow the exact steps in my [Setting up Jetson Xavier NX](https://jkjung-avt.github.io/setting-up-xavier-nx/) blog post, you should get similar results as mine.

jch-wang, 2020-10-13:
> I have no idea. If you start with a clean JetPack-4.4 installation of Jetson Xavier NX and follow the exact steps in my [Setting up Jetson Xavier NX](https://jkjung-avt.github.io/setting-up-xavier-nx/) blog post, you should get similar results as mine.

You may be right. My Jetson Xavier NX is not new, and I have run other Code Repositories on my Jetson Xavier NX before. This probably had an impact, though I don't know exactly what the impact was.

closed 2020-10-15 by jkjung-avt