jkjung-avt/tensorrt_demos #259: MTCNN Problem on Jetson NX and AJX not Detecting All Faces

AnasMK, 2020-10-13:
First, thank you, for this great repo.
We tried to run MTCNN on Jetson Nano, TX2, NX, and AJX. We tried 2 videos with different qualities 480, 720, 1080. We noticed that on Jetson Nano and TX2, it is detecting all faces and works fine on images and videos. but on NX and AJX has wired behavior.

On NX and AJX images works fine and able to detect all faces, the issue is when running MTCNN on videos. In the beginning, it detects all faces after number of frames, it was not able to detect all faces. the video has 8 faces in the middle, it was able to detect 3 faces then drop to one face only. Trying same video cropped and the 8 faces frame was in the beginning, it detected all the 8 faces.

a workaround we used is to keep initializing MTCNN after a certain number of frames. and it was able to detect all faces in the video but caused less FPS.

Is there something wrong with MTCNN on NX or AJX?

marlexct, 2020-10-14:
I am redoing the conversion from .h5 (pnet,rnet,onet) to engine for tensorRT. Now, I am in my way to reproduce the Cython build part, of your implementation. But I do not know what does part do. Please, could you help me with more information, I think in that step you are going to combine the three engines... Please, I would really want to learn, could you bring me some links in order to understand how are you doing those steps (to understand how to implement that by myself). I am using TensorRT 7.1.3.4.

jkjung-avt, 2020-10-14:
> We tried to run MTCNN on Jetson Nano, TX2, NX, and AJX. We tried 2 videos with different qualities 480, 720, 1080. We noticed that on Jetson Nano and TX2, it is detecting all faces and works fine on images and videos. but on NX and AJX has wired behavior.

@AnasMK  I have observed the problem in another way myself.  I tested my TensorRT MTCNN engines with [the Avengers poster](https://thumbor.forbes.com/thumbor/960x0/https%3A%2F%2Fblogs-images.forbes.com%2Fscottmendelson%2Ffiles%2F2019%2F03%2FAvengers-Chinese-Poster-D.jpg).  All tests were done on the same Jetson Nano DevKit.  With TensorRT 5 and 6 (earlier JetPack's), the TensorRT MTCNN demo could detect all faces (except Wong) in the poster.  But with TensorRT 7 (JetPack-4.4), it is _missing Captain America_ as shown below!  It would be very difficult to debug this problem.  I could only suspect it's **due to some changes in TensorRT 7** release for now...

[image]

AnasMK, 2020-10-14:
@jkjung-avt I have tested the same image on Jetson Nano and TX2 both have Jetpack 4.4 TRT 7,  gave me the same result as yours.

AnasMK, 2020-10-14:
@marlexct are you talking about this repo [tf2trt_with_onnx](https://github.com/riotu-lab/tf2trt_with_onnx), if you are please open issue on that repo, to keep each issue to the topic. I will be glad to help.

marlexct, 2020-10-14:
@AnasMK Nope, I am talking about @jkjung-avt work conversion of mtcnn (second step of demo 2 that is the third step of demo 1), I used your code (based on nvidia instructions, to convert h5 to engine for TRT). I would like more information in order to understand that step.

marlexct, 2020-10-14:
Hi, I have tested your code with DGX-A100 with TensorRT 7.1.3.4, it worked for me. I am sharing Avenger picture result.
[image]

jkjung-avt, 2020-10-16:
> Now, I am in my way to reproduce the Cython build part of your implementation (step 2 of mtcnn). But I do not know what does part do. Please, could you help me with more information, I think in that step you are going to combine the three engines...

@marlexct As explained in my [Running TensorRT Optimized GoogLeNet on Jetson Nano](https://jkjung-avt.github.io/tensorrt-googlenet/) post, **the Cython code is used to expose C++ functions so that I could call them from python**.

For example, I do [import pytrt](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mtcnn.py#L6) in "utils/mtcnn.py".  Then I do [pytrt.PyTrtMtcnn()](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mtcnn.py#L6) to [instantiate a TrtMtcnnDet (C++) class](https://github.com/jkjung-avt/tensorrt_demos/blob/master/pytrt.pyx#L62).  When I [call the forward() method](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mtcnn.py#L288), it gets translated to calling the C++ function [here](https://github.com/jkjung-avt/tensorrt_demos/blob/master/pytrt.pyx#L93) and [here](https://github.com/jkjung-avt/tensorrt_demos/blob/master/pytrt.pyx#L107) and the [TrtMtcnnDet::forward()](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trtNet.cpp#L259) in "trtNet.cpp" gets called eventually.

I keep the C++ and Cython code in this repo so that the GoogLeNet and MTCNN demos could work on older versions of TensorRT.  You could replace the forward() function with TensorRT python API.  Just make sure you handle inputs/outputs of your TensorRT engines correctly.

jkjung-avt, 2020-10-16:
> Hi, I have tested your code with DGX-A100 with TensorRT 7.1.3.4, it worked for me. I am sharing Avenger picture result.

Since number of GPU cores and GPU shared memory sizes differ a lot between Jetson Nano/TX2 and DGX-A100, TensorRT (even with the same version of 7.1.3.4) **would choose different optimized CUDA kernels (or different ways to fuse layers)** to implement layers in the PNet/RNet/ONet on the 2 kinds of platforms.

I'm still inclined to think _the originally reported problem_ lies in the TensorRT libraries...

marlexct, 2020-10-16:
> @marlexct As explained in my [Running TensorRT Optimized GoogLeNet on Jetson Nano](https://jkjung-avt.github.io/tensorrt-googlenet/) post, **the Cython code is used to expose C++ functions so that I [...]

Thank you so much for your explanation.

closed 2020-10-25 by jkjung-avt